arduino/Arduino #4142: LCD display bug in the latest version 1.6.6

AlexRH93, 2015-11-13:
Hola soy Alex Rodríguez, estudiante de robótica y me iniciado en la programación de micro controladores este curso lo que me parece un mundo realmente interesante.
El día 12/11/2015 con la versión del software 1.6.6 experimente un fallo con la pantalla LCD PCF8574T, ya que solo se mostraba un carácter en cada fila de la pantalla , en el ejemplo "Hello world", aparecía la "H" y la "w".
Al principio parecía un error de las librerías LiquidCrystal, pero cuando añadimos las nuevas tampoco funcionaba correctamente.

Al final decidimos probar el ejemplo en otro ordenador con la versión 1.6.5 del software y la misma placa y pantalla y funcionaba perfectamente, asín que estamos casi convencidos que es fallo de la versión 1.6.6. 

Hi I'm Alex Rodriguez, a robotics and I started programming in microcotroladores this course which I think is a really interesting world.
Yesterday the software version 1.6.6 develop a fault with PCF8574T LCD screen , it showed only one character in each row of the display , in the example "Hello world" , it appeared showed the "H "and" w " .
At first it seemed an error of LiquidCrystal librarie, but when we add the new didnt work properly.

In the end we decided to try the example on another computer with version 1.6.5 of the software and the same plate and screen and it worked perfectly , asynchronous we are almost convinced that it is version 1.6.6 phallus .

cmaglie, 2015-11-13:
Hi @AlexRH93 

this bug has been already reported, can you post a (possibly simple and small) sketch that shows the problem?
are you using `LiquidCrystal` or `LiquidCrystal_I2C`?
which board are you using?

cmaglie, 2015-11-13:
@AlexRH93 
I don't know if this may be related your problem:
https://github.com/marcoschwartz/LiquidCrystal_I2C/issues/5

serge123, 2015-11-14:
If have problem to use LiquidCrystal_I2C library with LCD - LCD does not display all symbols,
try to edit arduino core library .../hardware/arduino/avr/cores/arduino/Print.cpp, find:
==========================================================
/* default implementation: may be overridden */
size_t Print::write(const uint8_t *buffer, size_t size)
{
  size_t n = 0;
  while (size--) {
    if (write(*buffer++)) n++;
    else break;
  }
  return n;
}
==========================================================
Change it to:
==========================================================
/* default implementation: may be overridden */
size_t Print::write(const uint8_t *buffer, size_t size)
{
  size_t n = 0;
  while (size--) {
    n += write(*buffer++);
  }
  return n;
}
==========================================================

cmaglie, 2015-11-14:
@serge123 
instead of reverting `Print::write` it would be better to fix LiquidCrystal_I2C as I've already reported here:

https://github.com/marcoschwartz/LiquidCrystal_I2C/pull/5/files

```diff
 LiquidCrystal_I2C.cpp 
@@ -9,7 +9,7 @@
 #define printIIC(args)	Wire.write(args)
 inline size_t LiquidCrystal_I2C::write(uint8_t value) {
 	send(value, Rs);
-	return 0;
+	return 1;
 }
 
 #else 
```

Are you aware of other wrong implementations of LiquidCrystal_I2C (besides the one linked above)?

serge123, 2015-11-14:
@cmaglie 
No, I don't.
Thank you for LiquidCrystal_I2C.cpp fix. It also works for me.

sandeepmistry, 2016-03-09:
Closing as this has been reported upstream https://github.com/marcoschwartz/LiquidCrystal_I2C/pull/5.

closed 2016-03-09 by sandeepmistry

Biohidraulica, 2017-11-14:
Thank you, 

marcoschwartz/LiquidCrystal_I2C,  fix, and also works for me.