arduino/Arduino #4143: Really is good to have a Processing based IDE?

crushedice2000, 2015-11-13:
Hi! I think that the current IDE (that Processing based thing) offers a awful experience for power users.

Is there any milestone to make anything better?

Wouldn't be better to officially maintain a multiplatform package with the Arduino core and use the current IDE simply as interface?

----

Example of new/advanced user workflow:

    1.
            New: Download a platform-specific bundle with the Arduino core plus that awful IDE.
            Advanced: Download a platform-specific installer of the Arduino core.
    2.
            New: Start learning.
            Advanced: Install [Atom](http://atom.io) bindings for the Arduino core and start coding. 

---

Would be great to create a Atom package that includes the Arduino core and Atom bindings (adding a handy interface to build, upload, debug... along with a serial port monitor, a board selection menu, a menu for board-specific options, a library manager, a board manager...). This would be great. The current IDE has many improvements, but the font rendering, the syntax highlighting, the text editing, and the log viewing are really weird.

lmihalkovic, 2015-11-13:
The sophistication of the IDE is by no means a function of the underlying technology. The codebase for the current IDE is actually clean, [this issue](https://github.com/arduino/Arduino/issues/4084) is one of several attempts at showing how much can be done with a bit of cleanup in the editor, a bit of imagination and a bit of time. I have no doubt that the result can match, if not outpace any alternative you can think about (particularly when it comes to code completion).

ffissore, 2015-11-16:
Thank you for your feedback. Our web team is working on Arduino Create, a web based editor https://blog.arduino.cc/2015/05/05/sneak-peak-arduino-create/
However, I'm going to close this issue as it's neither a bug or a feature request for the current IDE

closed 2015-11-16 by ffissore

crushedice2000, 2015-11-16:
@ffissore: This seems a good project. However, I think that would be great to have a offline, multi-platform and fully functional Arduino core and tools (avrdude, avr-gcc, etc.). The web IDE is more for learning, but (for me) is a irking process to load a webpage to build a C++ file... ;-)

About the issue closing, I agree, this is only a suggestion and not a valid issue.
---

Off-topic questions:

1. Will this web IDE run offline (os-specific package with the web server and tools)?
2. How can a web browser open a serial port or use a AVRISP without depending on installed bindings?

crushedice2000, 2015-11-16:
@lmihalkovic: I didn't found this issue... Is really a good proposal, but... Is as ugly as the current IDE. I don't know how it behaves because is a tentative design only ;-)

mastrolinux, 2015-11-16:
@crushedice2000 there is an agent to speak with the boards from the  web, it is already open source: https://github.com/arduino/arduino-create-agent 

And we are investigating a way to run the IDE offline. The agent and the new Arduino compiler CLI allows already to create your own IDE on top of those technologies.

lmihalkovic, 2015-11-16:
@crushedice2000 I have to agree with you that the current IDE does not look particularly appealing. but I don't think it is a matter of technology at all. since I started playing with the ide code, I ran into the upcoming Arduino Create... I liked its fresh design, so I checked how much work would be involved in migrating the Java code. as I said, the code is mostly well written, and the migration straightforward... at the moment I am using the [dark theme](https://github.com/arduino/Arduino/issues/4158)

@mastrolinux pretty amazed that the company would bet all its resources for an entire year on a new IDE when so many things can so easily be addressed/improved in the current one. why not do both in parallel?

pedzed, 2017-04-15:
The current IDE just works, which is nice. However, for real programmers, this is quite a turn off. Why not build upon [Atom](https://atom.io)? You have [PlatformIO IDE](https://atom.io/packages/platformio-ide) as an Atom package, but it's a huge mess in my opinion. And besides, it would be nice to have an IDE backed by Arduino LLC.