arduino/Arduino #4145: Page setup font size and line numbers.

Sembazuru, 2015-11-14:
In addition to requesting code highlighting colors to show in printout (maybe as an option in the page setup) as mentioned in issue https://github.com/arduino/Arduino/issues/3013, I'd like to see choosing font size and and option to turn on (or off) line numbers in the printout.

I often print to pdf for carrying my source codes around and viewing them on platforms that I don't have the IDE/editor that I used to write the code installed. Having syntax highlighting and line numbers adds value to the printouts. I can (and have) load the .ino into a programmer's editor (PSPad in my case) to print using C/C++ highlighting, but that doesn't highlight the library keywords like the IDE does.

Because my display is a highDPI monitor (3200x1800 on a 15" screen) I have increased the font-size to 18pt for the editor. But this is way to large for printing. Please add a font-size setting in the page settings so I can reduced down to a reasonable size for printouts.

Roo426, 2016-01-04:
I have the same problem as Sembazuru, It would be nice to have a text editor font size and print font size.

Sembazuru, 2016-02-03:
As a side note, having to increase the font size for High-DPI monitors is currently being dealt with in https://github.com/arduino/Arduino/pull/4376
Printing still needs work. Not to extend the scope, but custom headers and footers wouldn't be amiss either...

Sembazuru, 2016-02-11:
cmaglie decided that fixing the font-size for printing actually isn't in scope with the general UI scaling for High-DPI monitors. Hopefully discussion and focus will come to this issue soon.

FritsBee, 2017-10-22:
Hi I use Notepad ++ for printing the *.ino in color