arduino/Arduino #4146: Cleanup the key handling code in the editor

lmihalkovic, 2015-11-14:
There are a number of open issues that can be traced to a few problems with the configuration of the code editor. This task is to address the core issue and *remove* the hacks. The cleanup will allow

* switching to the latest version of the RTA lib (i am using 3.xx - I also removed the patches to allow running the version directly from the distribution instead of having to patch things up for ArduinoIDE each time)
* fixing some of the navigation issues (nextTab/prevTab...)
* support for clean management of key bindings (#4086)

note: the situation is still controllable and the removal work is still not too overwhelming, but it is the right time to remove the cruft, and new features will come almost for free.

matthijskooijman, 2015-11-16:
With RTA you mean RsyntaxTextArea, I presume? AFAIK there is only a single change made to that library to help with the undomanager (see [this repository](https://github.com/arduino/RSyntaxTextArea/commits/2.5.8.1+arduino)). Are you saying this change is no longer needed or integrated in RTA master?

You talk about issues with key handling, can you be more specific about what issues you are referring to?

ffissore, 2015-11-16:
Once again, thank you @lmihalkovic However, you yourself mentioned the issues the already existing issues. Hence this issue is a duplicate. Moreover, we already use the latest version of RTA, which is version 2.5.8 https://github.com/bobbylight/RSyntaxTextArea/releases. And no, we won't use its master branch: we already have enough issue to deal with.

closed 2015-11-16 by ffissore

lmihalkovic, 2015-11-16:
@ffissore  I can imagine how difficult it seem to handle the maintenance. I hope that at some point you get to have the time and space to see that some of the open issues would go away by removing code, rather than adding more short term fixes.

NOTE: master works just fine.

lmihalkovic, 2015-11-20:
Cleaning up would be nice in order to support clean binding management, but it is possible to at least display things without too much headaches

[image]

lmihalkovic, 2015-11-26:
Done