arduino/Arduino #4148: Spelling error on arduino.cc

alextaujenis, 2015-11-14:
Go to this page: http://playground.arduino.cc/Main/Participate
The side-bar on the left has the word Partecipate when it should be spelled Participate.
[image]

per1234, 2015-11-14:
I've been trying to get this fixed for 2 months now. See https://github.com/arduino/Arduino/issues/3506#issuecomment-138879685. I have a list of 24 different issues with the website but I don't see the point of taking the time to report them if I'm just going to be ignored.

matthijskooijman, 2015-11-16:
The playground is essentially a wiki, created and maintained by the community, so anyone should be able to register an account and change it. However, I'm not so sure how to edit the side bar on the playground, perhaps it's a page with a special name?

per1234, 2015-11-16:
As I state in my issue, the page is http://playground.arduino.cc/Site/SideBar and it has a different password from the rest of the playground so the community can't edit that page. @mastrolinux has the ability to edit it. In fact he made this spelling mistake while making some of the changes I recommended in #3506.

matthijskooijman, 2015-11-16:
@per1234, ah, I should have read your issue before commenting here. I'm closing this issue as a duplicate of #3506, since that also includes this same issue. @alextaujenis, thanks for reporting, though!

closed 2015-11-16 by matthijskooijman

alextaujenis, 2015-11-16:
Looks like @mastrolinux fixed it, thanks!