arduino/Arduino #4150: "Bootloader file specified but missing" when bootloader file name contains expansions.

sleemanj, 2015-11-14:
I use variable expansions in the bootloader.file name in order to have combinations of sub menus select the booloader file,

    ...bootloader.file=optiboot_{build.mcu}_{build.f_cpu}_{upload.speed}.hex
    ...menu.clock.16MHz.build.f_cpu=16000000L
    ...menu.version.atmega8.build.mcu=atmega8

etc...

This works fine in 1.6.5-r5, in 1.6.6 it also works but emits a bogus warning when uploading using the bootloader (burning the bootloader is fine) as a step between compilation and upload.  It has no effect on the process, just is disconcerting for the user.

Abbreviated output below showing issue below.

[Board Manager package json](https://raw.githubusercontent.com/sleemanj/optiboot/master/dists/package_gogo_diy_atmega8_series_index.json)

1.6.6 (issues bogus warning)
------------------------------------------------------------------------------------------------------------------------

    "/usr/local/stow/arduino-1.6.6/hardware/tools/avr/bin/avr-objcopy" -O ihex -R .eeprom  "/tmp/buildd299ef9e4076492fa94192584887a770.tmp/Blink.ino.elf" "/tmp/buildd299ef9e4076492fa94192584887a770.tmp/Blink.ino.hex"
    Bootloader file specified but missing: /home/boffin/.arduino15/packages/diy_atmega8_series/hardware/avr/2015.11.13/bootloaders/optiboot_{build.mcu}_{build.f_cpu}_{upload.speed}.hex

    Sketch uses 1,030 bytes (3%) of program storage space. Maximum is 32,256 bytes.

1.6.5-r5 (no warning issued)
------------------------------------------------------------------------------------------------------------------------

    /usr/local/stow/arduino-1.6.5-r5/hardware/tools/avr/bin/avr-objcopy -O ihex -R .eeprom /tmp/build1581537743175209742.tmp/Blink.cpp.elf /tmp/build1581537743175209742.tmp/Blink.cpp.hex 

    Sketch uses 1,030 bytes (3%) of program storage space. Maximum is 32,256 bytes.

ffissore, 2015-11-16:
This issue was moved to arduino/arduino-builder#63

ffissore, 2015-11-16:
Thank you @sleemanj, working on it

closed 2015-11-16 by ffissore