arduino/Arduino #4151: Modernize the look of the UI to track Arduino Create

lmihalkovic, 2015-11-15:
Create looks really nice, and it would make sense to align the two UI where it makes sense. This is a first cut where only the buttons are still hard-coded (the rest is has been cleaned up and the images can be deleted).

[image]

TODO: 
* fix the remaining spacing issues
* restore the local popup menu

ffissore, 2015-11-16:
Thank you once more @lmihalkovic However, very few features of Create will be ported to the IDE

closed 2015-11-16 by ffissore

lmihalkovic, 2015-11-16:
It is quite unfortunate considering how little work is involved in modernizing the UI that nobody has ever taken one hour to do it. But I understand that as a company, it is preferable to do the bare minimum to just *maintain* an old product while all efforts are focussed on the next generation.

matthijskooijman, 2015-11-16:
> nobody has ever taken one hour to do it.

That seems like a pretty severe (and incorrect) assumption. Looking at improved syntax highlighting, error highlighting, a new library manager, etc. added in the last year or so, things *are* improving. It's just that all of these "one hours" can only be spent once, so choices must be made.

lmihalkovic, 2015-11-17:
I never questioned that the team is doing an awesome job, and I think we can all agree that it is unfortunate that there will never be any time for this. Arduino Create did all the design (really great job, btw!!). Implementation was just a matter of altering EHeader.paintComp(), adding 10 line to Themes (to avoid duplications between the themes, the base theme.txt is always applied, and an optional variation is applied after if the config says so), adding a couple lines to EToolbar, and creating a couple of new classes to draw the dot-logo and the new title box. I still have to replace the green buttons with java code, to make it work with any background color.  

But again i understand the decision to spend more efforts on the new product. Some companies still consider a unique visual identity an asset though.

lmihalkovic, 2015-12-04:
[LMDONE]

q2dg, 2016-09-06:
@lmihalkovic Could you do a pull request, please?