arduino/Arduino #4152: Arduino version 1.7.7 us already installd

dcarlson48, 2015-11-15:
I am running windows 10

I installed Arduino version 1.7.7 and then tried to run a sketch that I created in with arduino-1.6.5-r5-windows. which I believe ran with no issues. The errors were not related to the coding in the sketch. The appeared to be issues with the actual application. I thought it might have been an incomplete install so I uninstalled this install and tried to re- install. I received the below window. saying version 1.7.7 was already installed and the installation stopped.

[install error.docx](https://github.com/arduino/Arduino/files/34787/install.error.docx)

I then tried to install the older version but I got the same error. 

I went into the remove apps in the windows 10 settings window. I could not see the actual application but saw an Arduino SRL icon. I tried to uninstall this but it would not uninstall. 

Can anybody tell me how I can clean my system and get a working installation again?

Chris--A, 2015-11-15:
Arduino 1.7.7 is a badly versioned clone of this project provided by arduino srl.
You could ask them for advice for removing their product: https://github.com/arduino-org/Arduino

Also, you could try deleting the Arduino folder from 'program files', then run something like ccleaner to remove its registry entries.

To get a working version you can download the windows.zip from https://www.arduino.cc/en/Main/Software rather than using the installer version (.exe)

ffissore, 2015-11-16:
@dcarlson48 you're not using arduino.cc IDE. If you think it's worth, you can ask arduino.org to remove their fork, as many others did https://github.com/arduino-org/Arduino/issues/2

closed 2015-11-16 by ffissore