arduino/Arduino #4154: Third party support seems to still be missing sections on the Bootloader etc

rogerclarkmelbourne, 2015-11-15:
Third party spec page in the wiki does not document how to add new bootloaders

It currently reads..

```
Burn Bootloader

TODO...
```

Its a shame this doc has never been finished, and I can understand its very low priority, but reverse engineering how this works will take quite some time :-(

agdl, 2015-12-18:
@rogerclarkmelbourne I'm assigning this to @facchinm since he and @cmaglie are working on the new boot-loaders so they can write the wiki with the latest infos!

per1234, 2017-04-05:
Just wanted to clarify the wiki page referred to here is:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification
I agree it would be a very good idea to document this. It seems obvious to someone intimately familiar with the inner workings of the IDE but for anyone else some decent documentation can be extremely helpful. There's no need for this to wait for the completion of any bootloader updates. The documentation is independent of the bootloader version used.

per1234, 2020-09-30:
Fixed by https://github.com/arduino/arduino-cli/pull/925

The documentation is now here:
https://arduino.github.io/arduino-cli/dev/platform-specification/#upload-using-an-external-programmer
https://arduino.github.io/arduino-cli/dev/platform-specification/#burn-bootloader
https://arduino.github.io/arduino-cli/dev/platform-specification/#programmerstxt

@rogerclarkmelbourne if you find any of that to be confusing, incorrect, or incomplete, feel free to open an issue at https://github.com/arduino/arduino-cli/issues

closed 2020-09-30 by per1234