arduino/Arduino #4155: When external editor is set, hitting upload scrolls back to the top of file.

BlitzKraft, 2015-11-15:
I often compare files and have some example files open for reference. I lose my position in the file if I hit upload or verify. It would be better if the IDE doesn't scroll to the top and I can edit/compare the same line without having to scroll back down.

This happens exclusively when the external editor is set. When it is not set, it has the expected behavior of not scrolling, when I hit upload.

matthijskooijman, 2015-11-16:
I suspect this is because the content of the textarea is cleared and reloaded, resetting the scroll position. The fix would probably to save the current scroll position before reloading the file and restoring it afterwards.

If you're able to code this, a pullrequest would be helpful.

BlitzKraft, 2015-11-16:
I haven't worked with the IDE before. But I will look into that.

ffissore, 2015-11-17:
Fixed. Fix will be available in next hourly build http://www.arduino.cc/en/Main/Software#hourly

commit d0bd09c8 (2015-11-17): Saving caret position when using external editor, attempting to restore it on window activation. Fixes #4155
closed 2015-11-17 by ffissore