arduino/Arduino #4158: Support theme variants

lmihalkovic, 2015-11-15:
The upcoming Arduino Create looks awesome, and seems to support different color variants. This is about adding the same feature to Arduino IDE.

[image]

[image]

Note: at the moment the code is still using buttons.gif for the toolbar. It would be nice to replace them with either a simple color mask with the same content, or just code (like for the drop-down and the tabs)

lmihalkovic, 2015-11-15:
Credit for the look is entirely going to the [Arduino Create](https://blog.arduino.cc/2015/05/05/sneak-peak-arduino-create/) designers

ffissore, 2015-11-16:
Thank you @lmihalkovic. We won't add support for multiple themes. It's a feature you'll find in Create, not in the IDE

closed 2015-11-16 by ffissore

lmihalkovic, 2015-11-16:
No problem. It took a few minutes to do it, and i like it as it is.

lmihalkovic, 2015-11-17:
In case you change your mind, this is my theme-dark.txt

```
# GUI - TABS
header.bgcolor = #2c353a
header.text.selected.color = #abbdc1
header.text.unselected.color = #a2b5bb
header.tab.unselected.bgcolor= #384247
header.tab.selected.bgcolor= #171e20

# EDITOR - DETAILS

# foreground and background colors
#editor.fgcolor = #000000
editor.fgcolor = #707778
editor.bgcolor = #171e20

# GUI - BUTTONS
# LJPM delete when replaced with code
header.toolbar.buttons= buttons3.gif
```

lmihalkovic, 2015-11-26:
Done