arduino/Arduino #4162: Firmware for Wifi 101 shield is missing

q2dg, 2015-11-15:
In https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/firmwares only there is old Wifi shield firmware.

I've noted this reading this paragraph from https://www.arduino.cc/en/Main/ArduinoWiFiShield101:
*"The header marked as TX/RX allows the connection to the board's hardware serial, when shorted. **It can be used to upgrade the WiFi module FW** through the Arduino or Genuino board. *

SurfingDude, 2015-11-18:
The WiFi 101 board shipped with firmware version 19.3.0 which is an Atmel product. The object code of the firmware can be found as an Atmel Studio Example project, WINC1500 Firmware Update Project for the SAMD21Xplained Pro. It seems pretty mature but there could always be updates coming for it. I don't know if this is Atmel proprietary or if it's something that users could mess with.

q2dg, 2015-11-18:
Thanks. That's the question...If the final answer is that Wifi101 shield's firmware is propietary (what a shame!), then I wouldn't mention in doc the possibility of upgrading it, because it confuses about what can or can't be done. If the answer is that Wifi101 shield's firmware is open, then I would add a page explaining the process (as it was with older Wifi shield).

cmaglie, 2015-11-26:
The documention has been updated. The firmware is proprietary, it may be eventually upgraded if Atmel release a new version, we are setting up a procedure, but it doesn't require the use of TX/RX pairs.

q2dg, 2015-11-26:
Ok!
I'll close this issue, then.
Thanks!

P.D: Firmware is proprietary...mmmm....mmmm...mmmm

closed 2015-11-26 by q2dg

cmaglie, 2015-11-26:
> P.D: Firmware is proprietary...mmmm....mmmm...mmmm

I know... I hope that Atmel decide to release the source code of the firmware but we cannot force them to do it.
On the other hand there are a lot of RF modules with closed-source firmware (Xbee, BTLE), so it's not so "uncommon" this kind of setup.