arduino/Arduino #4164: Create different search contributors for the global search

lmihalkovic, 2015-11-16:
The global search in #4077 should be extensible. At the moment there are 3 default contributors. It would be interesting to know what kind of information people would like to have.

Possible examples:

* list of tutorials available on arduino,cc  (keyword "tutorials"). Selecting one should directly open the matching page. Would be nice if the company would have a json file somewhere with that list, instead of having to do some scraping
* list of boards ( keyword "boards"). Again, selecting an element should open a *official* description page
* list of events.. a quick way to access a list of *sanctioned* events registered with arduino.cc

The list of possibilities is endless...

ffissore, 2015-11-16:
Since the feature doesn't yet exist, it doesn't make much sense opening another issue for saying what else it should do. Please keep the conversation going in #4077

closed 2015-11-16 by ffissore

lmihalkovic, 2015-12-04:
[LMDONE]