arduino/Arduino #4166: Reference Update regarding F() macro

oldmanegan, 2015-11-16:
I believe an explanation of the use of the F() macro should be included in the Reference. Not sure if it should be listed under Utilities or what. The fact that it is located far down on the PROGMEM page means that most people may not ever find this and be stumped on reducing SRAM demand of their programs where they use Serial.print as a debug tool. I believe two things ought to take place:
1. Move the F() macro text from PROGMEM to the Reference section, say under Utilities
2. Add a link to this text at the top of the PROGMEM section with a small note of what it does (for example: "NB - if you are trying to reduce SRAM demand while using Serial.print, please check out the F() macro here <add link>."
Would surely help many new to Arduino.

agdl, 2015-12-17:
@oldmanegan I'm sorry but it cannot be moved in the main reference page, because the structure listed there it's let's say "fixed from the high level", if you want you can contribute improving what is already online in the actual progmem reference page and saying where it can be useful to link this page!

agdl, 2016-01-04:
I'm closing this, reopen it if needed please!

closed 2016-01-04 by agdl