arduino/Arduino #4168: Linux installer fails creating a desktop shortcut because it uses a hardcoded path to desktop

jonnius, 2015-11-16:
When running install.h in Arduino 1.6.6 Linux, an error occurs:

    cp: cannot create regular file '/home/jonatan/Desktop/arduino.desktop': No such file or directory

It fails on localized systems where the Desktop's path is different. The correct path can be found in the variable XDG_DESKTOP_DIR in ~/.config/user-dirs.dirs.

ffissore, 2015-11-17:
#3226 contains an effort for improving linux install script. Can you please verify the proposed script solves your issue?

jonnius, 2015-11-17:
The script does not solve this issue, yet.

jonnius, 2015-11-21:
The script solves the issue now.

NicoHood, 2015-11-22:
This should not be closed until the PR is merged. I really like to see this PR merged though. @ffissore I think its time now, please merge it :)

ffissore, 2015-11-23:
Fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-11-23 by ffissore