arduino/Arduino #4174: sprintf / snprintf error

nuket, 2015-11-18:
There's an error in `sprintf` and `snprintf` that occurs when more than 8 varargs are passed in after the format specifier.

The problem is that:
* an extra '00' sneaks into the output string between the 1st and 2nd parameter outputs
* the last parameter does not make it into the output string

This can lead to day-long debugging sessions of unrelated software modules, in order to determine the root cause.

Here's an example sketch that demonstrates what happens:
```
/**
 * Demonstrate that the sprintf / snprintf implementation on Arduino is broken.
 * 
 * sprintf / snprintf, when used with more than 8 varargs:
 *     inserts an extra '00' in the output string
 *     removes the last parameter from the output string
 * 
 * This causes a lot of extra debugging.
 * 
 * Expected: abcd0001020304050607
 * Actual:   abcd0000010203040506
 */

enum {
    OUTPUT_LENGTH = 50
};

uint32_t shortId   = 0xabcd;
uint8_t  buffer[8] = { 0, 1, 2, 3, 4, 5, 6, 7 };

void setup() 
{
    Serial.begin(115200);
    
    char output[OUTPUT_LENGTH] = { 0 };

    // ERROR: The buggy way.
    //
    // Prints: abcd0000010203040506
    snprintf(output, OUTPUT_LENGTH, "%04x%02x%02x%02x%02x%02x%02x%02x%02x",
        shortId,
        buffer[0],
        buffer[1],
        buffer[2],
        buffer[3],
        buffer[4],
        buffer[5],
        buffer[6],
        buffer[7],
        buffer[8]);

    Serial.println(output);

    // OK: The way that actually works.
    //
    // Prints: abcd0001020304050607
    snprintf(output, OUTPUT_LENGTH, "%04x",
        shortId);

    snprintf(output + 4, OUTPUT_LENGTH - 4, "%02x%02x%02x%02x%02x%02x%02x%02x",
        buffer[0],
        buffer[1],
        buffer[2],
        buffer[3],
        buffer[4],
        buffer[5],
        buffer[6],
        buffer[7],
        buffer[8]);

    Serial.println(output);

    // ERROR: Another buggy way.
    //
    // Prints: abcd0000010203040506
    sprintf(output, "%04x%02x%02x%02x%02x%02x%02x%02x%02x",
        shortId,
        buffer[0],
        buffer[1],
        buffer[2],
        buffer[3],
        buffer[4],
        buffer[5],
        buffer[6],
        buffer[7],
        buffer[8]);

    Serial.println(output);
}

void loop() 
{
}
```

matthijskooijman, 2015-11-18:
Actually, I think the problem is that you are passing a uint32_t (4 bytes) argument for a %x (hexadecimal int) format. %x expects an int, which is 2 bytes on AVR. Since printf isn't particularly type-safe, it will take the first 2 bytes of your `shortId` for the %04x and then the second 2 bytes (which will be 0x0000) for the first %02x. Try using %04lx (or %08lx if you really want a full 4-byte number), where "l" indicates "long".

I'm closing this issue, as I think my above analysis points out the error is in your code. If you disagree, feel free to add a comment and we can reopen (but be sure to actually post a testcase that shows adding one more identical argument breaks it, instead of one differently typed argument).

closed 2015-11-18 by matthijskooijman

nuket, 2015-11-18:
Your analysis was correct. Even after all these years, C likes to surprise me. I actually did want to print a short, so I ended up doing a `(uint16_t)` cast on the `shortId` in the parameter list instead.

However, after further digging, it [became clear](https://vilimpoc.org/blog/2015/11/18/sprintf-snprintf-problem-on-arduino/) that the Arduino IDE suppresses all of the compiler warnings that would have immediately indicated where the problem was coming from. This totally violates [the principle of least astonishment](https://en.wikipedia.org/wiki/Principle_of_least_astonishment) and I'm pretty sure I'm not the first one to get bitten by this.

I'd argue that the Arduino IDE should ship with `-Wall` on by default, instead of `-w` (which is crazy).

matthijskooijman, 2015-11-18:
In the latest IDE versions, warnings can be enabled in the preferences. Changing the default is still the plan, but I'm not sure about the timeframe (ideally, warnings are only enabled in the sketch, to prevent surprising novice users with warnings in library code that they didn't write, but that feature wasn't written yet). Perhaps you could create an issue to request warnings to be enabled by default, at least for the sketch?

nuket, 2015-11-18:
Done.