arduino/Arduino #4176: Problem with OSX and Arduino 1.6.7: Unable to verify/upload code

mishelg, 2015-11-18:
Have a Yosemite 10.5 Macbook pro with JDK 8. Building the files work fine but the IDE can not upload/verify.
[image]

ffissore, 2015-11-18:
Can you please turn on Verbose compilation (File > Preferences) and paste the new compilation output?

mishelg, 2015-11-18:
Hi here is the verbose output I get while trying to compile code shown in the above image.

[exec] /Users/mishel/Arduino/build/macosx/work/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Users/mishel/Arduino/build/macosx/work/Arduino.app/Contents/Java/hardware" -tools "/Users/mishel/Arduino/build/macosx/work/Arduino.app/Contents/Java/tools-builder" -tools "/Users/mishel/Arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/avr" -built-in-libraries "/Users/mishel/Arduino/build/macosx/work/Arduino.app/Contents/Java/libraries" -libraries "/var/root/sketchbook/libraries" -fqbn=arduino:avr:uno -ide-version=10607 -build-path "/var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/T/build8ded6a0e498170c7399f876312690e8f.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/T/untitled2090063220.tmp/sketch_nov18a/sketch_nov18a.ino"
     [exec] /Users/mishel/Arduino/build/macosx/work/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Users/mishel/Arduino/build/macosx/work/Arduino.app/Contents/Java/hardware" -tools "/Users/mishel/Arduino/build/macosx/work/Arduino.app/Contents/Java/tools-builder" -tools "/Users/mishel/Arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/avr" -built-in-libraries "/Users/mishel/Arduino/build/macosx/work/Arduino.app/Contents/Java/libraries" -libraries "/var/root/sketchbook/libraries" -fqbn=arduino:avr:uno -ide-version=10607 -build-path "/var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/T/build8ded6a0e498170c7399f876312690e8f.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/T/untitled2090063220.tmp/sketch_nov18a/sketch_nov18a.ino"
     [exec] "/Users/mishel/Arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Users/mishel/Arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Users/mishel/Arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/T/build8ded6a0e498170c7399f876312690e8f.tmp/sketch/sketch_nov18a.ino.cpp" -o "/var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/T/build8ded6a0e498170c7399f876312690e8f.tmp/preproc/sketch_nov18a.ino.cpp_5577006791947779410_preprocessed.cpp"
     [exec] "/Users/mishel/Arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Users/mishel/Arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Users/mishel/Arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/T/build8ded6a0e498170c7399f876312690e8f.tmp/sketch/sketch_nov18a.ino.cpp" -o "/var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/T/build8ded6a0e498170c7399f876312690e8f.tmp/preproc/sketch_nov18a.ino.cpp_8674665223082153551_preprocessed.cpp"
     [exec] "/Users/mishel/Arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Users/mishel/Arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Users/mishel/Arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/T/build8ded6a0e498170c7399f876312690e8f.tmp/sketch/sketch_nov18a.ino.cpp" -o "/var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/T/build8ded6a0e498170c7399f876312690e8f.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"
     [exec] "/Users/mishel/Arduino/build/macosx/work/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino4/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/T/build8ded6a0e498170c7399f876312690e8f.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"
     [exec] exit status 1
     [exec] Error compiling.

ffissore, 2015-11-19:
Ok now please do the following. Try again to compile, with verbose output on. Latest command line printed before dying is `ctags` call (`/Users/mishel/Arduino/build/macosx/work/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino4/ctags`). **While keeping the IDE open**, copy paste that line into a new terminal

Copy paste here the whole output shown in that terminal

tmsanrinsha, 2015-12-20:
I met the same problem and called ctags in command line
```
 /Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino5/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/var/folders/8s/2xqcrr5j45dgtkx3jmnc3clr0000gn/T/build48f25a83dfc94bc42b80ddad99d957e3.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"
ctags: append mode is not compatible with tags to stdout
````
In my .ctags, --append=yes is written. This option is overridden by calling ctags with --append=no option.

```
 /Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino5/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives --append=no "/var/folders/8s/2xqcrr5j45dgtkx3jmnc3clr0000gn/T/build48f25a83dfc94bc42b80ddad99d957e3.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"
```

matthijskooijman, 2015-12-20:
Hm, so we should make sure that ctags doesn't read from any existing configuration files. Looking at the manpage of exuberant-ctags, adding `--options=NONE` as the *first argument* to ctags achieves this. I'm not sure if this also works for the version used by Arduino, though.

tmsanrinsha, 2015-12-21:
In my environment, it works
```
/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino5/ctags --options=NONE -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/var/folders/8s/2xqcrr5j45dgtkx3jmnc3clr0000gn/T/build48f25a83dfc94bc42b80ddad99d957e3.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"
```

dekay, 2015-12-31:
I had this same problem with 1.6.7 under Arch Linux.  Changing "--append" to "--append=no" in ~/.ctags fixed it for me.

sej7278, 2016-01-02:
i'm getting something similar with 1.6.7 on debian - first build seems to work, then if i rebuild, i get:

```
build-esp8266/preproc/ctags_target_for_gcc_minus_e.cpp.o: In function `HardwareSerial::begin(unsigned long)':

HardwareSerial.h:77: multiple definition of `setup'
```

and also ```multiple definition of 'loop' ``` errors.

matthijskooijman, 2016-01-04:
@sej7278, this looks like a completely different problem. Could you please open a new issue for your problem? Also make sure to include the full compiler output (of both the first and second build) with verbose compilation enabled.

agdl, 2016-07-12:
I'm closing this for lack of feedback. Please reopen this if needed

closed 2016-07-12 by agdl