arduino/Arduino #4180: Serial Plotter Labels

agdl, 2015-11-18:
As discussed in https://github.com/arduino/Arduino/pull/4022 I think could be a great improvement to have some kind of label in the new multi-lines serial plotter. My suggestion is to have something like:
```arduino
Serial.print("i:");
Serial.print(i);
Serial.print(" ");
Serial.print("j:");
Serial.println(j);
```

To identify the labels

ffissore, 2015-11-18:
/cc @henningpohl

agdl, 2015-11-18:
Furthermore, if you print let's say 4 data you have a legend with 5 colors

mobluse, 2018-01-01:
I think a better system to set labels is to use a line with headings. The current system is like a CSV file and then one could continue in that style. CSV is a [standard](https://www.ietf.org/rfc/rfc4180.txt). If you want to update the legend you send out a new header. A header is distinct from data since it contains fields that can't be converted to numbers.

Your format is acceptable to me though, and the data simulator for serial in https://makecode.microbit.org/ uses it.

Maybe one could have both systems, since it is possible to distinguish them because of ":" in one of the systems.

chromhelm, 2018-04-05:
Hi
I implement the label functionality. Have a look.
https://github.com/arduino/Arduino/pull/7422

sandeepmistry, 2019-09-16:
This has been added in https://github.com/arduino/Arduino/pull/7453.

closed 2019-09-16 by sandeepmistry