arduino/Arduino #4182: Switch Case Anomalies/Bugs

oldmanegan, 2015-11-18:
Guys,
I searched and did not find this replicated elsewhere so here are my observations (for Arduino IDE 1.6.6:
1. The switch case cannot be nested without causing issues. 
2. As well, it was found that if statements inside case statements may be unpredictable. 
3. As well, the for statement using ‘0’ as first entity value does not work, need to be ‘1’ based, then use decrement in the then statements to get to the zero element.
4. Cannot define a variable inside a case statement.

cmaglie, 2015-11-18:
For points 1. 2. and 3. looks like bugs in your code than anomalies of the compiler, have you placed `break` after each case?
http://www.drpaulcarter.com/cs/common-c-errors.php#2.1

>  4. Cannot define a variable inside a case statement.
you can't, you must add instructions in curly braces `{` to do that `}`.

If the above doesn't solve your problem please post a sketch that shows the problem.

oldmanegan, 2015-11-18:
Breaks are in each case statement.
I will post code when I can get to it.
If curly braces are required, then the Reference should state this explicitly. Please pass this on to the Reference people.
Best Regards,

John
johnvegan@gmail.com

agdl, 2016-07-12:
Reference updated to point this out.

closed 2016-07-12 by agdl