arduino/Arduino #4183: Board Manager problem with installing the tools

keestux, 2015-11-18:
Hey,

We have our own JSON for our boards. It's a SAMD board which uses the same tools as the Zero. After selecting the install it downloads all the necessary tools. But then there is a conflict between where the tools are installed and where Arduino IDE expects them.

Our name is SODAQ and the Board Managers installs the tools in $HOME/.arduino15/packages/SODAQ/tools/. So far so good. Installation reported no problem. Then I select our board, and build a sketch. The error I get is:
```
Cannot run program "{runtime.tools.arm-none-eabi-gcc.path}/bin/arm-none-eabi-g++" (in directory "."): error=2, No such file or directory
```

However when I install the Zero, (which installs the very same tools in $HOME/.arduino15/packages/arduino/tools/) it all works.

So, either the installer should have installed it in $HOME/.arduino15/packages/arduino/tools/, or the IDE should assign {runtime.tools.arm-none-eabi-gcc.path} to my SODAQ/tools.

In case you want to test, here is the URL (used for testing)
```
http://downloads.sodaq.net/test/package_sodaq_index.json
```

aethaniel, 2015-11-19:
@keestux, from what I understood, you should not need to fill the "tools" section in your json file as you already filled the "toolsDependencies" to reference the ones provides by Arduino.

Can you test by putting:
```
      "tools":
      [
      ]
```

ffissore, 2015-11-19:
I can't replicate this with latest hourly, even though I see your json file is now listing all tools and I've removed them in my test

keestux, 2015-11-19:
@aethaniel That is something I already tried. And it doesn't make a difference. To be sure I tried it again, and with the same conclusion. The IDE installs the tools at packages/SODAQ/tools. But then if you build a sketch it looks in packages/arduino/tools.

@ffissore Yes, you're right, and also 1.6.6 is working. Hmm, I _did_ try 1.6.6 this week but it gave errors from arduino-builder, and I couldn't figure out what went wrong.

Anyways, I'm happy now. We shall instruct our users to use 1.6.6 or higher. Thanks

BTW It is still a waste of disk space that the tools are installed twice (276Mb) in my ~/.arduino15 tree. Once for the SODAQ boards and once for the Zero.

ffissore, 2015-11-20:
@keestux yes, sorry about that. We'll eventually fix it

ffissore, 2015-11-20:
We are solving it. Follow ups in #4193

ffissore, 2015-11-20:
Fixed. Fix will be available in next hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-11-20 by ffissore

keestux, 2015-11-20:
Thanks. You're too kind :-)