arduino/Arduino #4185: HardwareSerial and SoftwareSerial should have a common super class

drmpf, 2015-11-19:
HardwareSerial and SoftwareSerial are supposed to be variations on a Serial connection, but they do not share a common superclass

Please add a superclass,  'Serial_", or some such, which contains the methods common to both HardwareSerial and SoftwareSerial.

At present libraries cannot accept HardwareSerial / SoftwareSerial objects interchangeably.

Particular use case is a library to determine the baud rate of an attached device.  
Need to call multiple begin() methods with different baud rates, but have to have lots of extra code to workout if user is using HardwareSerial or SoftwareSerial for the connection.
On UNO, most likely has to use SoftwareSerial.
On Mega can/should use HardwareSerial

hicklin-james, 2016-04-06:
Any progress on this as of yet? I am attempting to write a library for the ESP8266 wifi module and would like it's constructor to accept either a `SoftwareSerial` or `HardwareSerial` object. I can get most of the way there using the `Stream` superclass, but unfortunately it doesn't implement the `begin` function.

NicoHood, 2016-04-06:
I dont think this is a good idea. This adds just more overhead. And you will not use softserial and hardwareserial at the same time. If you really want to support both of them a template would be the better choice or simply two different constructors.

Also in my opinion softserial is unreliable in most cases
https://github.com/arduino/Arduino/issues/4822