arduino/Arduino #4187: IDE doesn't show core package specific libraries

aethaniel, 2015-11-19:
I have tested 2 different Board Manager packages containing a library with only Out-of-the-Box examples with same result:

```
Invalid library found in C:\Users\pippopluto\AppData\Local\Arduino15\packages\AMEL\hardware\samd\1.0.0\libraries\SME_basic
```

A quick discussion with @cmaglie may lead the issue is due to lack of source code because the library contains only specific examples in .ino form. This would make sense regarding how Arduino libraries are formalized.

Is it an issue or are we trying to use an IDE mechanism not planned for that?
If latter case, would it be possible and easy to allow Board Manager Packages to embed specific examples?
I think mainly about having a formalized 'examples' folder at package root, ie same level as also optional 'libraries' folder.

Thanks,

T.

cmaglie, 2015-11-19:
> Is it an issue or are we trying to use an IDE mechanism not planned for that?

exactly, I think that the current check is correct (a library without source code is not a library) and the proper fix for this is to add support for examples in cores.

per1234, 2015-11-19:
The workaround is to put a dummy .h file in the library folder. See https://github.com/NicoHood/HoodLoader2/commit/4d95de6d1cc3b31b196661f75f270dd26e79b42b for an example.

per1234, 2015-11-19:
I have submitted a pull request to the AMEL-Tech Boards repository to add the dummy file: https://github.com/ameltech/sme-arduino-core/pull/9
@aethaniel if you tell me which other Boards Manager package has this issue I'll attempt to do the same for it also.

aethaniel, 2015-11-19:
hi @per1234, I was just coming back here to write adding a dummy .hpp with same name as the library one works ok.
I can update both repos https://github.com/ameltech/sme-arduino-core/ and mine here: https://github.com/AtmelUniversityFrance/atmel-avr-xmini-boardmanagermodule.

I will merge your PR for SME right now and add the file in my own branch for Atmel AVR Xplained minis. I consider this as a temporary workaround.

By the way, I would clearly prefer for that particular purpose to be allowed having an 'examples' folder at package root, as this doesn't fit the 'library' concept.
@ffissore, what is your opinion about that?

ffissore, 2015-11-20:
/cc @00alis for UX feedback

per1234, 2017-07-04:
Closing as duplicate of https://github.com/arduino/Arduino/issues/3297

closed 2017-07-04 by per1234