arduino/Arduino #4189: [Library Manager] Updatable libraries

agdl, 2015-11-19:
If a core has in bundle let's say the Ethernet library with a version that is < of the latest version of the Arduino one, you are bounced between the latest version of the library of the added core and the one before the last of the Arduino core.

@ffissore @cmaglie

cmaglie, 2015-11-19:
What's your expected behaviour?

Maybe a duplicate https://github.com/arduino/Arduino/issues/4064 ?

agdl, 2015-11-19:
yep duplicate. I'm closing this

per1234, 2015-11-27:
@agdl are you referring to the issue of the library always appearing in the Type: Updatable section of the Arduino IDE 1.6.6 Library Manager no matter what you do. If so,  this isn't a duplicate of #4064, it's a separate issue, but can be easily fixed by changing the name value in the library.properties of the bundled library to be different from the Arduino IDE Built-In library. See: https://github.com/esp8266/Arduino/pull/1096 and https://github.com/JChristensen/mighty-1284p/pull/23

VioletGiraffe, 2018-01-10:
Guys, the SD library bug is not critical but seriously annoying, can someone be done about this, please?

per1234, 2021-05-14:
Closing as fixed by https://github.com/arduino/Arduino/commit/ac6d3c1afffd81d429498b9d00bc2c92f849cbfe

Prior to that commit, Arduino Library Manager used the installed versions of the [platform bundled libraries](https://arduino.github.io/arduino-cli/latest/platform-specification/#platform-bundled-libraries) as a factor in determining whether an update was available. In addition to not being helpful, since platform bundled libraries are not distributed by Library Manager, but instead by Boards Manager, it also caused this peculiar "always updatable" bug under certain conditions. The change to making Library Manager ignore platform bundled libraries also eliminated this bug.

closed 2021-05-14 by per1234