arduino/Arduino #4190: git master. Build process fails to download some .zip files

JeanLucCoulon, 2015-11-19:
Hi,

When I start from a clean git tree, and I want to build arduino, the build process fails to download the following files:
Bridge-1.1.0.zip   Keyboard-1.0.0.zip
Robot_Motor-1.0.2.zip
Esplora-1.0.4.zip  Mouse-1.0.0.zip
SpacebrewYun-1.0.0.zip
Robot_Control-1.0.2.zip
Temboo-1.1.4.zip
Firmata-2.5.0.zip
RobotIRremote-1.0.2.zip

The error message is:

untar-unzip-download:
      [get] Getting: https://github.com/arduino-libraries/Firmata/archive/2.5.0.zip
      [get] To: /usr/local/src/arduino-cc/arduino-1.6.7/build/Firmata-2.5.0.zip
      [get] Error getting https://github.com/arduino-libraries/Firmata/archive/2.5.0.zip to /usr/local/src/arduino-cc/arduino-1.6.7/build/Firmata-2.5.0.zip

untar-unzip-checksum:
     [echo] Testing checksum of "Firmata-2.5.0.zip"
 [checksum] Could not find file /usr/local/src/arduino-cc/arduino-1.6.7/build/Firmata-2.5.0.zip to generate checksum for.

BUILD FAILED

If I download the file "by hand" and rebuild, it goes a bit further and fails on the next missing one.

I've downloaded all the indicated file myself and everything went fine.

Regards

Jean-Luc

ffissore, 2015-11-20:
I've just run a clean build and it built flawlessly. Could it be a network error? Maybe github was down for a couple of minutes or your network was down?

JeanLucCoulon, 2015-11-20:
Hi,

No, it is not.
The offending files arent in the github repository.
They are downloaded at build time.
But once download, you can build arduino, even if there are changes in the tree at the condition the changes are not for these files.

I've this problem since several weeks now.

This morning once again because Bridge-1.6.0.zip has changed its version.
So I've downloaded it directly with the browser. And everything was fine.

I've set the "ignoreerrors" parameter to "true" in untar-unzip-download:
<get src="${archive_url}" dest="${archive_file}" verbose="true" ignoreerrors="false" />

And I got the following:
usr/local/src/arduino-cc/arduino-1.6.7/build/build.xml:85: The following error occurred while executing this line:
/usr/local/src/arduino-cc/arduino-1.6.7/build/build.xml:544: The following error occurred while executing this line:
/usr/local/src/arduino-cc/arduino-1.6.7/build/build.xml:179: The following error occurred while executing this line:
/usr/local/src/arduino-cc/arduino-1.6.7/build/build.xml:188: The following error occurred while executing this line:
/usr/local/src/arduino-cc/arduino-1.6.7/build/build.xml:206: The following error occurred while executing this line:
/usr/local/src/arduino-cc/arduino-1.6.7/build/build.xml:664: javax.net.ssl.SSLException: java.lang.RuntimeException: Unexpected error: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty

However,I don't know the value the trustAnchors parameter should have.

Regards

Jean-Luc

ffissore, 2015-11-20:
It looks like your JDK is unable to connect to github because of SSL certificate issues. Double check you're using JDK 1.8. Here is mine `java -version`
```
federico@smilzo:~$ java -version
openjdk version "1.8.0_45-internal"
OpenJDK Runtime Environment (build 1.8.0_45-internal-b14)
OpenJDK 64-Bit Server VM (build 25.45-b02, mixed mode)
```

JeanLucCoulon, 2015-11-20:
Seems to be similar (from debian sid):

[jean-luc@tangerine] % java -version                                          ~
openjdk version "1.8.0_72-internal"
OpenJDK Runtime Environment (build 1.8.0_72-internal-b05)
OpenJDK 64-Bit Server VM (build 25.72-b05, mixed mode)

I've even tried version 1.9. With the same problem.
With 1.7, it doenst build.

J-L

ffissore, 2015-11-20:
1.7 won't work: IDE requires 1.8. I'm sorry but I'm out of options. Is your system up to date? Are root SSL certificates up to date?

JeanLucCoulon, 2015-11-20:
Yes yes, my system is up to date. Debian sid with some parts of experimental.

I've no idea of which SSL certificate can be needed for that.

probonopd, 2016-01-23:
Fedora 23:

```
[root@host build]# java -version
java version "1.8.0_60"
Java(TM) SE Runtime Environment (build 1.8.0_60-b27)
Java HotSpot(TM) 64-Bit Server VM (build 25.60-b23, mixed mode)
```

Seems to run into a similar issue:

```

unzip:
     [echo] Unzipping SpacebrewYun-1.0.0.zip into folder linux/work/libraries

untar-unzip-check:

untar-unzip-download:
      [get] Getting: https://github.com/temboo/Temboo/archive/1.1.4.zip
      [get] To: /home/me/Arduino/build/Temboo-1.1.4.zip
      [get] Error opening connection java.io.FileNotFoundException: https://github.com/temboo/Temboo/archive/1.1.4.zip
      [get] Error opening connection java.io.FileNotFoundException: https://github.com/temboo/Temboo/archive/1.1.4.zip
      [get] Error opening connection java.io.FileNotFoundException: https://github.com/temboo/Temboo/archive/1.1.4.zip
      [get] Can't get https://github.com/temboo/Temboo/archive/1.1.4.zip to /home/me/Arduino/build/Temboo-1.1.4.zip

untar-unzip-checksum:
     [echo] Testing checksum of "Temboo-1.1.4.zip"
 [checksum] Could not find file /home/me/Arduino/build/Temboo-1.1.4.zip to generate checksum for.

BUILD FAILED
/home/me/Arduino/build/build.xml:93: The following error occurred while executing this line:
/home/me/Arduino/build/build.xml:87: The following error occurred while executing this line:
/home/me/Arduino/build/build.xml:569: The following error occurred while executing this line:
/home/me/Arduino/build/build.xml:188: The following error occurred while executing this line:
/home/me/Arduino/build/build.xml:203: The following error occurred while executing this line:
/home/me/Arduino/build/build.xml:215: The following error occurred while executing this line:
/home/me/Arduino/build/build.xml:713: Could not find file /home/me/Arduino/build/Temboo-1.1.4.zip to generate checksum for.
```

However, the Temboo URL seems to be entirely invalid and leads to a 404.

probonopd, 2016-01-23:
Strange, this ticket now says "ffissore was unassigned by probonopd 2 minutes ago" but I have done nothing of this sort (intentionally) and I don't even think I'd have the rights to assign/unassign someone.

Bogusmips, 2016-01-23:
Obviously same behaviour than probonopd on Debian Jessie because the Temboo url lead to a 404.

cmaglie, 2016-01-26:
the problem with Temboo is different see: #4473

sandeepmistry, 2019-09-16:
Based on the comments in https://github.com/arduino/Arduino/issues/4473, this should be good now, but please re-open if you still face any issues.

closed 2019-09-16 by sandeepmistry