arduino/Arduino #4194: Add Multi plot examples

pierrotm777, 2015-11-20:
Can you add examples for the multi plot feature ? 

Thanks

per1234, 2017-07-04:
There is one here:
https://www.arduino.cc/en/Tutorial/ArduinoSoundSpectrumSerialPlotter
but it doesn't make it clear that's what it is. It looks like there was an intention of adding a comment explaining the purpose of the delimiter print:
```arduino
Serial.print("\t"); // 
```
but it didn't happen. I wasn't even sure whether that would work as the author of the feature had only [mentioned ](https://github.com/arduino/Arduino/issues/3451#issuecomment-151124346)commas or spaces as valid delimiters.

I think it would be more useful to have a bundled example that demonstrates the feature in a way that is obvious and accessible to everyone rather than something based on such specific hardware, as in the ArduinoSoundSpectrumSerialPlotter tutorial.

per1234, 2020-09-25:
Moved to https://github.com/arduino/arduino-examples/issues/7

closed 2020-09-25 by per1234