arduino/Arduino #4195: Library Manager: display Arduino Libraries at the beginning of the list

00alis, 2015-11-20:
@ffissore When I open the Library Manager the Type dropdown filter is correctly set on All, we should just display Arduino Libraries at the beginning of the list, and all the others afterwards in alphabetical order.

@mbanzi

lmihalkovic, 2015-11-20:
If you might consider that some sort of quick search could make sense, then a possible way to use it could be to display that same information. Could all be done via keyboard by typing "CTRL+3", "libs" ('l' is enough). More differentiation in the display could include color, font, .... just a thought.

[image]

ffissore, 2015-11-23:
Fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

commit ceef83db (2015-11-23): Libraries that contain type "Arduino" are now listed at the top. Fixes #4195
closed 2015-11-23 by ffissore