arduino/Arduino #4196: Consider eliminating buttons.gif

lmihalkovic, 2015-11-20:
The idea is to replace them with java code

q2dg, 2015-11-21:
Why?

lmihalkovic, 2015-11-21:
IMHO, this is a small nuisance (I was struggling with changing the back-color for the dark theme because of the anti-aliasing) that once out of the way leaves some interesting doors open. I don't feel qualified to go into a debate as to which and why, but I just wanted to bring the thought to a place of debate. I do apologize if that is an inconcenience.

[NTS]

jesusbravo38, 2015-11-21:
Please remove me from this email threads

closed 2015-11-22 by lmihalkovic

matthijskooijman, 2015-11-23:
@jesusbravo38, you probably clicked "watch" in the Arduino repo at https://github.com/arduino Unsubscribing is something only you can do, through your github account, so we can't help you there.

@lmihalkovic, your issues really lack motiviation and clarity. In this case, I'm not even sure what you're asking. Are you suggesting to encode a bitmap using java code that draws lines and points? That doesn't really sound like something that is maintainable or easy. You say buttons.gif is a "nuisance", but don't explain what this means or how it is a nuisance.

Reporting issues like these will only waste our time and result in them being closed. Please be more thoughtful and complete when reporting issues. If you are not prepared to invest time in properly explaining or motivating things, it might be better to not report them at all (or pick one or two issues that your feel are most important, and direct the available time to those).

lmihalkovic, 2015-11-23:
@matthijskooijman I see your point, and you may have noticed that **I closed the issue myself** as I was not prepared to go into a lengthy argument about why and how I thought the gif file was in the way. You will also no doubt have noticed that unlike some of my other suggestions (many are of the form "add xxx", "add yyyy"), this one was formulated as "CONSIDERATION", allowing an early dismissal without opening if there was zero interest for the CONSIDERATION. 

As for the rational I didn't feel would be of any use, IMHO the presence of this image creates an arbitrary point of contention preventing
* the extension of different parts of the UI via the Tools extension mechanism (look into the archives for confirmation)
* the skinning of the app cleanly

I replaced the images with vector graphics code (need a better designer than me), which took care of my skinning issue. To address the maintenance issue, I generate the code from SVG files that I create in a graphical editor.

For the other contention point, I introduced a contribute-able menu class that would allow people who know what they want/how to do what they want to contribute their *Tool* extension to different parts of the UI. This is **not the reason** for having written the code, but this is a almost-free side-effect of its existence (I use it to populate the outline view with different contribution types depending on the kind of file that is selected: main sketch, image, doc, lib, ... ).

Once again, this was offered as food for thoughts. Again, I **closed it myself** to signify that although I did see enough value to have created it, I was also keenly aware that I might be alone down that path. Should a like minded traveller have said "hey... yeah.. you're there too?! lets ... " I would have happily
expanded as I just did.

Hope this clarifies the situation.

lmihalkovic, 2015-11-23:
@matthijskooijman using svg2java (somewhere here on ghub) on svgs produced as an export from affinity designer (mac). Results are ... good enough to validate the workflow, but will need some tweaking to become pixel perfect. I suspect it is my ignorance that is at fault (improper scaling during export).

matthijskooijman, 2015-11-24:
@lmihalkovic Thanks for clarifying, I have a much better picture of what you're suggesting now. Incidentally, I've been doing some work that required adding toolbar buttons and also found that the toolbar code isn't particular flexible, partly because all buttons are in a single file.

If you have code to render buttons from SVG, and/or apply other cleanups, I'm certainly interested in that. Perhaps you could make a pullrequest, or share your code in a branch on github? Even if you do not have time to provide a proper, ready-to-merge pullrequest, I might have some time to use your code as a base for cleaning up the toolbar.

lmihalkovic, 2015-11-25:
This is most of the code (still a couple of things to fix, like the hardcoded minWidth):  https://gist.github.com/lmihalkovic/999e75a8e32730160680
I took away to dependency on Editor (the actions base class works the normal swing way for dealing with contextual dependencies - see TextAction and JTextArea for a 'clean' example - I don't go via AppContext like they do to get the same result, but what is a minor detail).

The list of actions is passed via constructor. at the moment the list is hardcoded, will likely be replaced with discovery via ServiceLoader (java standard).

Icon base classes are code generated from SVGs which are exported from this newfangled drawing app I bought - still haven't figured how to do the export properly (i.e. normalized 0/1.0 coordinates). workflow tip: I draw them with stupid color (like RED, BLACK, BLUE) so that when I get the generated Java code, all I have to do is replace setBackground(**new Color(255,0,0)**) with bg(). bg()/fg()/... are default methods on a ArduinoIcon interface. All they do it Theme.get("asdkjfhasdkfh") to get the right color. so... maintenance of the draw code is dead trivial

lmihalkovic, 2015-11-26:
@matthijskooijman  assuming that by now you have done your cleanup, so i deleted the gist (code was obsolete and still filled with some of the original leftovers). I'm also done with gracefully stopping the compiler, but i am missing an icon for it. guess i will make something like the stop button in processing...

matthijskooijman, 2015-11-26:
Actually, I hadn't gotten around to looking at the gist yet, so I'm still interested in having it put up again.

lmihalkovic, 2015-12-04:
[LMDONE]

cmaglie, 2015-12-04:
@lmihalkovic 
did you realize that every message you send here is received by 680+ people?
may you find a less invasive way to track your work?

Thanks!

lmihalkovic, 2015-12-04:
you guys have limited resources and doing a great job with the hardware, cores, and Create... how about calling this the 2.0 that your resources don't allow you to work on yet? it does many of the things people have been asking for, has a cleaned up codebase, and some extras

[image]