arduino/Arduino #4197: Add 'support' for additional file types in a sketch

lmihalkovic, 2015-11-21:
I noticed that Arduino Create will support storing some sort of a README as well as a one or more images describing the sketch. I would be nice for the IDE to provide the same level of support for them.

In my case this extends to allowing some of these in libraries as well.

lmihalkovic, 2015-11-22:
Adding support for layout details if they are present in a sketch (i.e. any images in lib or sketch can be displayed inside the editor)

[image]

ffissore, 2015-11-23:
This has been internally discussed and there are no plans to add such features to the Arduino IDE

closed 2015-11-23 by ffissore