arduino/Arduino #4200: OTA on ESP-201

AdySan, 2015-11-22:
I've been using [this](https://github.com/arduino/Arduino/pull/4107) build for a few days with NodeMCU 0.9 hardware and it works flawlessly. Trying a ESP-201 today, I downloaded the BasicOTA example over serial. When I try to do an OTA download, I keep getting OTA_BEGIN_ERROR ([ERROR]: No response from device). I'm selecting Generic ESP8266 Module, Serial, 80MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck on 192.168.1.24. (GPIO15 -> GND, GPIO0 -> DTR, CH_PD -> 3.3V)

me-no-dev, 2015-11-22:
you are running out of flash space. 512k is not enough to fit both the running and the new sketch on the flash at the same time.

ffissore, 2015-11-23:
Please keep the conversation going on #4107

closed 2015-11-23 by ffissore