arduino/Arduino #4201: Add #include "Stream.h" to cores\arduino\USBAPI.h

MichaelJonker, 2015-11-22:
Hello,
I propose to add `#include "Stream.h"` to the header file `hardware\arduino\avr\cores\arduino\USBAPI.h` (e.g. following the `#include "Arduino.h" `), as such the USBAPI.h can be included without relying on Stream.h being included already before.

Motivation:
Currently Stream.h, which is required by USAPI.h, is imported by HardwareSerial.h, which itself is #included by Arduino.h before the `#include "USBAPI.h"`. If the effective inclusion of HardwareSerial.h is disabled (which can be achieved by a #define HardwareSerial_h before inclusion of Arduino.h) then Stream.h is not included leading to an error in the USBAPI.h header.

Context:
When using library HardwareSerialRS485, the loading of HardwareSerial is inhibited with a #define HardwareSerial_h. Until now this #define could be included in the HardwareSerialRS485.h header file (which also has a `#include "Stream.h"`). As of IDE 1.6.6, the `#include "Arduino.h"` has been moved to the very first line of the sketch, invalidating this option.

Hence, as an alternative solution, the board.txt mechanism will be used to define RS485 aware boards and adding lines like `Arduino_Uno_RS485.compiler.cpp.extra_flags=-DHardwareSerial_h`.
This works fine and the loading of HardwareSerial is properly blocked. However, on Atmel32U4 based devices, where USBAPI.h is loaded, the absence of Stream.h is now causing a problem. (Which, for the time being, has to be solved with a not so elegant construction).

Thanks for your attention

Michael Jonker

matthijskooijman, 2015-11-24:
Making includes between header files explicit is always a good idea to me, so including Stream.h from USBAPI.h sounds like a fine change to me. If you have made and tested the change already, perhaps you can submit a pullrequest? Should be easy to review and merge, I expect.

MichaelJonker, 2015-11-24:
Hoi Matthijs,

I have tested the change locally (i.e. adding `#include "Stream.h"` just after the `#include "Arduino.h"` in USBAPI.h). It allows me to elegantly solve a problem caused by the move of the `#include "Arduino.h"` to the very first line of the sketch introduced by IDE 1.6.6. (BTW, p have a look at my contributed HardwareSerialRS485 library at https://github.com/MichaelJonker/HardwareSerialRS485 ).

Adding `#include "Stream.h"` to USBAPI.h will not break existing code as a) currently this header is already included before the inclusion of USBAPI.h and b) Stream.h is guarded by a `#if defined(Stream_h)` construct.

Now to submit a pull request is really a lot of hassle for adding a single line. If you can point me to a light weight procedure, which does not involve forking and fetching the full ~2 Gbyte Arduino repo with all embedded history (I lost my local copy of the fork after a computer crash), I will be happy to do so.
(and I then might even change the picture on my profile, which currently reflects my feeling on using github ;-).

MichaelJonker, 2015-11-25:
Contrary to what I said above, the move of Arduino.h to the beginning of the sketch has some more implications.... To solve this, including Stream.h in USBAPI.h is not sufficient. However, it will certainly not harm and still be a good idea.

When I have time I can submit a pull request for a more complete correction.

matthijskooijman, 2015-11-25:
@MichaelJonker, you can edit individual files through the github web interface and create a pullrequest from that. You'll still need to create a fork in your github account, but there is no need to make a local clone. Please do make sure to add a proper comment message, the default "Changing some_file.cpp" or whatever it is really isn't sufficient :-)