arduino/Arduino #4202: Generated function prototypes for const __FlashStringHelper are incorrect.

etherfi, 2015-11-22:
Example code:

<pre>
#include <WString.h>

void setup() {
  Serial.print(test(true));
}

void loop() {}

const __FlashStringHelper* test(bool temp) {
  if (temp)
    return F("x");
  return F("y");
}
</pre>

Generated code in sketch_nov22a.ino.cpp:

<pre>
#include <Arduino.h>
#line 1
#line 1 "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_4790fd94ec03044aef00d007e6734c0b\\sketch_nov22a.ino"
#include <WString.h>

#line 3 "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_4790fd94ec03044aef00d007e6734c0b\\sketch_nov22a.ino"
void setup();
#line 7 "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_4790fd94ec03044aef00d007e6734c0b\\sketch_nov22a.ino"
void loop();
#line 9 "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_4790fd94ec03044aef00d007e6734c0b\\sketch_nov22a.ino"
const__FlashStringHelper* test(bool temp);
#line 3
void setup() {
  Serial.print(test(true));
}

void loop() {}

const __FlashStringHelper* test(bool temp) {
  if (temp)
    return F("x");
  return F("y");
}
</pre>

Note the function prototype "const__FlashStringHelper* test(bool temp);" is missing the space after "const".

As expected, it fails to compile:

<pre>
sketch_nov22a:9: error: 'const__FlashStringHelper' does not name a type

 const __FlashStringHelper* test(bool temp) {
 ^

C:\Users\taylorc\AppData\Local\Temp\arduino_4790fd94ec03044aef00d007e6734c0b\sketch_nov22a.ino: In function 'void setup()':

sketch_nov22a:4: error: 'test' was not declared in this scope

   Serial.print(test(true));
                         ^

exit status 1
'const__FlashStringHelper' does not name a type
</pre>

per1234, 2015-11-22:
Duplicate of https://github.com/arduino/arduino-builder/issues/69

ffissore, 2015-11-23:
This issue was moved to arduino/arduino-builder#71

closed 2015-11-23 by ffissore

ffissore, 2015-11-23:
Fixed. Fix will be available in next hourly build http://www.arduino.cc/en/Main/Software#hourly