arduino/Arduino #4205: initVariant() function not being linked in 1.6.6

kmclifton, 2015-11-23:
We have some custom init code that needs to run to initialise our hardware (not Arduino).  In version 1.6.5 of the IDE we simply included a file called initvariant.cpp (defining void initVariant() ) alongside pins_arduino.h under variants/standard in our package directory.  See contents of initvariant.cpp below.  With the upgrade to 1.6.6 that seems to have broken.  The init code no longer runs at all.  It isn't linked in.
Possibly related to this issue: https://github.com/arduino/arduino-builder/commit/aee32ff374a4b659af604f8ae461b3248bb51f65

```
#include <Wire.h>
#include <Arduino.h>
#include <pins_arduino.h>

void initVariant() 
{ 
	Wire.begin();
	pinMode(EXPANDER_RSTN, OUTPUT);
	digitalWrite(EXPANDER_RSTN, LOW);
	delay(10);
	digitalWrite(EXPANDER_RSTN, HIGH);

	delay(10);
}
```

ffissore, 2015-11-23:
This issue was moved to arduino/arduino-builder#72

closed 2015-11-23 by ffissore

ffissore, 2015-11-23:
Please keep the conversation going at https://github.com/arduino/arduino-builder/issues/72#issuecomment-158878239