arduino/Arduino #4207: Replacing the LiquidCrystal libary with NewLiquidCrystal

crushedice2000, 2015-11-23:
I think that LiquidCrystal should be replaced with NewLiquidCrystal (The fmalpartida LiquidCrystal fork) It's really great.

PaoloP74, 2015-11-26:
Easy...
Fork project, switch to GitHub platform and add to library manager.

bperrybap, 2015-11-29:
Even though I contributed to this library, I would not support replacing the standard LiquidCrystal library with fm's library. While fm's library is good, it is quite complex and is multiheaded (has multiple libraries bundled into a single library). It does have some issue related to this - like you can end up with "dead wood" or unused code being linked into your final image or end up in a situation were modules that are not even needed for a particular sketch won't compile in certain environments.

crushedice2000, 2015-11-30:
@bperrybap: Ok, I reviewed the code and you seem to have reason.

crushedice2000, 2015-11-30:
@PaoloP74: How can I maintain this project into the library manager?

PaoloP74, 2015-11-30:
--> https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

bperrybap, 2015-11-30:
fm's library was designed to be a replacement for the IDE supplied LiquidCrystal library - I was involved with steering things that way. However,  there are many issues that come up when libraries have collisions due to the way the IDE handles libraries and does it searches.
There are some cases that can cause the IDE to create incorrect include paths to create a scenario where it picks the wrong library directory so the code will not compile - when it could have had it picked the correct library order.
(I actually run into them quite often during my library development)
And while I like the concept of being able to replace a IDE supplied library like LiquidCrystal with a new one, it doesn't reliably work in practice in this case because fm's library is multiheaded and contains more than a single "library". So now not only can there be collisions issues with LiquidCrystal but there can and often is collisions with other libraries and classes like LiquidCrystal_I2C.
I see MANY threads on the forums where users screw up their libraries when trying to get their LCD i2c backpacks working. It is all user error and often from them trying to install more than one i2c LCD library and then having odd issues do to library collisions.
I see this and help users with this probably each week.

For less technical users, it is not obvious that a library named "LiquidCrystal" can break or cause collisions with a library named "LiquidCrytal_I2C". 
And since the IDE does not have a way to remove an installed library, it isn't easy for a user to try a library and remove it when it doesn't work or causes issues.
Because of this,  less technical users come running to the forums for support since they can't easily remove the library to get their IDE working again.
The real issue goes back to the way the IDE "looks" for libraries and handles the include path, but that is not going to be fixed any time soon.

For this reason, I think that the original approach of trying to have a single "library" that replaces an IDE supplied library using the same name along with multiple other device library classes is not a good approach.
To avoid conflicts it is better to have multiple libraries that separate out the classes.

From an administrative point, for the time being, I'd like to discourage the Arduino IDE admins/maintainers allowing any person or entity to add their 3rd party library to the library manager with the same name as an IDE supplied library. i.e. in this case "LiquidCrystal". 
But not just specific to this library. I'm meaning *any* IDE supplied library, Wire, SoftSerial, Servo, etc...

I think there are already enough issues related to IDE library updates and collisions that need to be dealt with without this added complication.
And that tossing the ability for 3rd parties to be able to add their own libraries to the library manager that replace IDE supplied libraries adds in a whole new set of complexity and issues.

Once the IDE is corrected to have the IDE do its updates for its supplied libraries down in the IDE install area vs stomping on the users 3rd party installed libraries in the {sektchbook}/libraries, and the IDE can uninstall a library, I think the restriction could be relaxed.
The discussion about the IDE updates and where supplied libraries are done is currently happening on the developers list and in other git issues.

lmihalkovic, 2015-11-30:
@bperrybap 
```
Once the IDE is corrected to have the IDE do its updates
```
What I sad about this whole lib story is that the answer has been around for 20 years in small projects like windows, unix/linux or even eclipse. But who knows, it could be that the Arduino requirements were so different that established solutions couldn't apply when the lib manager was coded

bperrybap, 2015-11-30:
@lmihalkovic 
I couldn't agree more. In fact the mystery of re-inventing the world extends into many parts of Arduino.

cmaglie, 2015-11-30:
> From an administrative point, for the time being, I'd like to discourage the Arduino IDE admins/maintainers allowing any person or entity to add their 3rd party library to the library manager with the same name as an IDE supplied library. i.e. in this case "LiquidCrystal". 

@bperrybap 
nobody was about to try to add a 3rd party library named `LiquidCrystal`, because we already have it in the library manager and is maintained by Arduino: https://github.com/arduino-libraries/LiquidCrystal
(actually the github organization "arduino-libraries" contains **all** the bundled libraries so they are basically "taken" and cannot be added to the library manager by 3rd parties)

> And while I like the concept of being able to replace a IDE supplied library like LiquidCrystal with a new one, it doesn't reliably work in practice in this case because fm's library is multiheaded [...]

I don't know the details of this library, I suppose that it provides a `LiquidCrystal.h` or a `LiquidCrystal_I2C.h` files, so that creates the ambiguity? There is a link to the repository?

bperrybap, 2015-12-01:
Here is the link:
https://bitbucket.org/fmalpartida/new-liquidcrystal/wiki/Home

It is multi layer library that consists of multiple  classes including:
- fast pin i/o
- i2cio for pcf8574 pin i/o 
- 4 bit "LiquidCrystal" 
- pcf8574 i2c interface with configurable wiring
- 2 or 3 wire/pin interface using a shift register
- 2 wire/pin interface using a shift register
-  3 wire/pin interface using a a shift register with configurable wiring
- 1 wire interface using a shift register
- byvac LCD interface

Each layer is its own class.
All of these are bundled together and installed as a single LiquidCrystal "library" from a single zip image.
It is essentially multiple libraries bundled together as one.

The Arduino recommend way of doing this would be to create a separate library for each class and install them separately.
There are other libraries that do it that way like some of the AdaFruit libraries.

Keep in mind that this library was done a while back  before there was even a way to install a library zip file from the IDE much less get the library and do updates from the network.
At that point in time, it was much easier for the user to install if it was bundled.

When lumped together, all the modules are compiled since that is how the IDE builds "libraries".
Things can get messy because of sub library dependencies like for Wire, and the included i2cio and fastio classes.

But sub library dependencies are messy in general because of the way the IDE does its builds and handles its libraries.
I really wish that it was done differently.
Even the new stuff recently added to the IDE to search for library dependencies within libraries still has issues.

I'll spare the further details but that is the summary of why fm's LiquidCrystal library is complicated.

as a FYI
I am looking at releasing some new 3rd party hd44780 LCD libraries with similar functionality as fm's library; however, they won't be all bundled together but instead will be separate libraries for each class.

@cmaglie 
Your comment about a library being "taken" is interesting.
So does that mean that you cannot have more than a single library added to the library manager with the same name?
If so this seems problematic as there are several libraries out there that have name collisions and I can envision a scenario where multiple 3rd parties all want to register their library.
To me this should be allowed if it isn't.
That way a user can select between which library they want.
But to do that, the library manager probably needs to be able to uninstall a library so a user could install another by the same name.

closed 2021-03-28 by per1234