arduino/Arduino #4208: Driver Install for Genuino Zero in 1.6.6

SurfingDude, 2015-11-23:
Apparently the genuino zero exists, but the edbg port driver for it is not in the drivers/genuino.inf file. Is this on accident or on purpose? I can't debug as I only can get arduino zero in my country. Discussion found on:
https://forum.arduino.cc/index.php?topic=360798.0

cmaglie, 2015-11-24:
The driver for Genuino Zero is here:
https://github.com/arduino/Arduino/blob/master/build/windows/dist/drivers/genuino.inf#L65

there is a genuino.inf/.cat separated from arduino.inf/.cat

SurfingDude, 2015-11-24:
OK, I'm sure it works for you, but it doesn't work for user amerlagha in topic 360798.0

Can I ask you to comment in that topic as to the status of this, and whether he has a bad .inf file or a defective genuine zero?

I posted it for you because I saw that arduino.inf contains three lines:
%zero.edbg.name%=DriverInstall, USB\VID_03EB&PID_2157&MI_01
%zero.sketch.name%=DriverInstall, USB\VID_2341&PID_804D&MI_00
%zero.bootloader.name%=DriverInstall, USB\VID_2341&PID_004D

While genuine.inf even in 1.6.6 contains only two:
%zero.sketch.name%=DriverInstall, USB\VID_2341&PID_824D&MI_00
%zero.bootloader.name%=DriverInstall, USB\VID_2341&PID_024D