arduino/Arduino #4214: Processing Graph sketch doesn't work in Processing version 3.0

tigoe, 2015-11-25:
The Processing sketch that comes with the Communications examples doesn't work in Processing 3.0, because you can't draw in SerialEvent() in Processing anymore (Thanks Eric Forman for catching this). Below an updated version of that sketch. 

The other Processing sketches included in the Communications examples should be checked too, to see that they run in 3.0.

```

// Graphing sketch

// This program takes ASCII-encoded strings
// from the serial port at 9600 baud and graphs them. It expects values in the
// range 0 to 1023, followed by a newline, or newline and carriage return

// Created 20 Apr 2005
// Updated 24 Nov 2015
// by Tom Igoe
// This example code is in the public domain.

import processing.serial.*;

Serial myPort;        // The serial port
int xPos = 1;         // horizontal position of the graph
float inByte = 0;

void setup () {
  // set the window size:
  size(400, 300);

  // List all the available serial ports
  // if using Processing 2.1 or later, use Serial.printArray()
  println(Serial.list());

  // I know that the first port in the serial list on my mac
  // is always my  Arduino, so I open Serial.list()[0].
  // Open whatever port is the one you're using.
  myPort = new Serial(this, Serial.list()[0], 9600);

  // don't generate a serialEvent() unless you get a newline character:
  myPort.bufferUntil('\n');

  // set inital background:
  background(0);
}
void draw () {
  // draw the line:
  stroke(127, 34, 255);
  line(xPos, height, xPos, height - inByte);

  // at the edge of the screen, go back to the beginning:
  if (xPos >= width) {
    xPos = 0;
    background(0);
  } else {
    // increment the horizontal position:
    xPos++;
  }
}

void serialEvent (Serial myPort) {
  // get the ASCII string:
  String inString = myPort.readStringUntil('\n');

  if (inString != null) {
    // trim off any whitespace:
    inString = trim(inString);
    // convert to an int and map to the screen height:
    inByte = float(inString);
    println(inByte);
    inByte = map(inByte, 0, 1023, 0, height);
  }
}

```

agdl, 2015-11-26:
@tigoe for what concern the other examples:
- dimmer works;
- physical pixel works;
- virtual color mixer works

agdl, 2015-11-26:
Thank you for the report. I'm closing this!

closed 2015-11-26 by agdl

tigoe, 2015-11-26:
Thanks!  The CallResponse examples should be clean, since they draw only in draw(). I think that's all the sketches that have Processing sketches with them,