arduino/Arduino #4215: Banner re-size issue on Launching IDE

sh-aks, 2015-11-25:
It starts with this image
[image]

after a few seconds
[image]

then IDE opens.

ffissore, 2015-11-25:
Resize is expected, but should be an instant. Does it last for longer? You have camstudio: can you record a video?

sh-aks, 2015-11-25:
It takes a bit longer. Change happening is visible.

Link to the video:
https://drive.google.com/file/d/0B9E0W6W6Y7p5Zml1YWNKUzBuNzg/view?usp=sharing

ffissore, 2015-11-25:
I see. Thank you very much. I'll try to make it a little more reactive

ffissore, 2015-11-25:
@sh-aks can you try latest hourly? I've updated the component we use to make .exe files (that the one showing the first, slightly larger splash screen), although I fear this won't fix your issue

sh-aks, 2015-11-25:
@ffissore : The Problem is still persistent in the latest hourly .

*Hourly still comes in a folder called "arduino-nightly".

ffissore, 2015-11-25:
Yep, we didn't change the file name. 

I've just pushed a patch for launch4j, which is the tool we use to make .exe files. It has a delay of 1 sec, now it's 100ms. Can you download the windows hourly and try again? http://www.arduino.cc/en/Main/Software#hourly

sh-aks, 2015-11-25:
@ffissore : The problem is still there in this build as well.

Would you need another video?

ffissore, 2015-11-25:
yes pls

sh-aks, 2015-11-25:
Video:
https://drive.google.com/file/d/0B9E0W6W6Y7p5RWxSNG1zMTRMUW8/view?usp=sharing

ffissore, 2015-11-26:
Ok, my patch doesn't work because launch4j comes with its own precompiled .o files, so patching source .c files is pointless. I've sent the patch upstream https://sourceforge.net/p/launch4j/git/merge-requests/7/. Hopefully it will get merged and a new launch4j will be released

CurveShot, 2015-12-30:
Hi, I might be having a problem similar to this one, however my banner is significantly smaller, and leads to a tiny IDE.

There are some screenshots in the forum post I mentioned 

http://forum.arduino.cc/index.php?topic=368650.0

cmaglie, 2016-02-11:
@sh-aks @CurveShot 
Can you test the latest hourly build? I think that this problem has been solved (together with a bunch of other problems) with #4376, here some references:

https://github.com/arduino/Arduino/pull/4376#issuecomment-171374850
https://github.com/cmaglie/Arduino/commit/885a51709938ccba2c317f155591e8df52a17059
https://github.com/arduino/Arduino/pull/4376#issuecomment-175102003

CurveShot, 2016-02-12:
Looks good @cmaglie

closed 2016-03-10 by cmaglie

sh-aks, 2016-03-17:
@cmaglie: Solved for me too in 1.6.8

Apologies for delay.

commit c3fc56f8 (2015-11-25): Windows: patched launch4j to reduce the delay between pollings for the main
window. Fixes #4215