arduino/Arduino #4218: closing IDE with x or Alt+F4 not working correct

willie68, 2015-11-25:
So, it seems to be, that i be the only windows guy here.
This issue is there since Sep 2013 (BUG #1592) , but no one (exl. Paul Stoffregen) is hearing me or reading the original description. 
So, i give this another (my last) try. 
Closing IDE Window with mouse click on x or with Alt+F4 won't save last edited file and window position. 
CTRL-Q is working, but for me unusual.
last tested version 1.6.6

lmihalkovic, 2015-11-25:
I am working on fork where i will address many of these irritations. I'm sorry It is in closed source for the moment but it will be addressed on way or another (hopefully in the main version before i do).

matthijskooijman, 2015-11-25:
>  I'm sorry It is in closed source

Note that the GPL license used by Arduino does not allow this. If you distribute the Arduino IDE in compiled form, you are obliged to also distribute, or at least offer the corresponding source code. I don't think this should be discussed further in this issue though, but I did want to mention it.

As for the actual issue, this sounds like a reasonable request, one would expect things to work the same regardless of how a window is closed. However, I suspect that there is a difference between closing the last window, which results in closing the application, and closing the application, which results in saving all window states and closing all windows. This is mostly a technical difference that shouldn't really influence the user experience, but it does influence the implementation, which is probably why the difference is present in the first place.

Consider what happens if you have two windows open, and press alt-f4 twice. A user might expect both windows to be re-opened when the IDE is restarted, though that's significantly more complex to implement (I believe) than only re-opening the last window. In any case, I would suggest that only re-opening the last window is sufficient.

@lmihalkovic, if you're interested in fixing this issue and providing a pullrequest, I'm pretty certain this should be small and useful enough to be merged quickly (provided the implementation is of sufficient quality, of course).

lmihalkovic, 2015-11-25:
@matthijskooijman  yes, thank you, I am keenly aware of the terms of the GPL. Even though you chose to quote a portion of my statement, I am sure you actually read it until the end (I'm sorry It is in **closed source for the moment**)

```
As for the actual issue, this sounds like a reasonable request, one would expect things to work the same regardless of how a window is closed
```
I have now enough experience with the codebase to know that seemingly obvious expectations like the one in question are far from a given. The code is full of inconsistencies when it comes to dealing with actions. But please, do *not take my word for it*, and look for yourself at all the menus.

This issue is by all means small. I am sure more time has been spent collectively debating it than it would have taken any of us to address it. As I said I will eventually address it if it is still pending when it becomes my next priority. My current priority is to fix the entanglement between consoles and provide a clean way to abort a compilation.

Quality like many things is in the eyes of the beholder... and as I said somewhere else, there exist a number of issues that are purely related to incomplete short-term fixes for other issues. To save us both a round-trip here is the example you will ask for: EditorListener.java: some issues must have existed leading to the conclusion that the class was needed as part of a solution. This led to more ongoing issues with shortcuts. I got rid of the class which fixed many shortcuts issues, as well as most of the menu inconsistencies.

lmihalkovic, 2015-11-25:
was there ever an icon designed for the "stop" button?! it is actually not even necessary to replace the ConsoleOutputStream to get everything to work.

[image]

willie68, 2015-11-26:
I think, first step should be open the last closed window. I don't use multiple windows in the arduino ide. And if i compare to other windows programs i use, no other will do so. They open the last state in **one** window, all tabs and so on, but they do not open 2 windows. Because this implifies, that you have start the program twice. (I know, that this is not correct with the arduino ide, but normal users on windows will believe this.)
Another point, which i found very problematic on my workshops, is that file/open opens another window. That's on windows unusual, too. "Open" in windows nomally means, open this file in the actual window. (Maybe as a new tab, if the program support this.) And here we have the point when the user thinks that the program was startet twice. This behavier is not windows like.
Look at some other ide's, like eclipse, netbeans, idea and how they deal with this problem. (Ok, i must admit, i'm a java guy) They have a project view, where the different projects live and the source frame is always the same.

willie68, 2016-01-05:
Just checked against 1.6.7. Not working...

willie68, 2016-01-05:
I have no one unassigned.

commit c98c1808 (2016-01-25): Save window position when closing app from "x" button or Alt+F4

Fix #4404
Fix #4218
closed 2016-01-25 by cmaglie