arduino/Arduino #4228: Switching Tabs broken

NicoHood, 2015-11-29:
Ctrl Alt Right switches 2 tabs for me. If I hold it for a few seconds the IDE switches like crazy. Afterwards I cannot type anything in the text editor again.

I also was confused how to even get to the tabs on the right. Scrolling over all tabs would make most sense, that is what I intuitive tried. As an alternative simple buttons would be also acceptable.

Example sketch:
https://github.com/watterott/HDMI-Display/tree/master/software/HDMI-Display

per1234, 2015-11-29:
If you click the down arrow(tabs menu) on the right of the tabs bar it starts working correctly.

matthijskooijman, 2015-12-09:
For ctrl+alt+left/right switching two tabs, it turns out that the key is handled both by EditorListener, and registered as a shortcut key on the next tab / prev tab menu items in the tab bar menu. Fixing that should be a matter of removing the EditorListener handling, I'll include this in the cleanup PR I'm preparing.

As for the "cannot type" thing, I cannot reproduce this here. I see the "switching like crazy" part, where the title updates continuously, the selected tab updates more slowly and once you let go of the key combo, it still continues to switch a while longer. With my cleaned up code, this crazy behaviour seems to be gone, especially when the duplicate keybinding is removed.

Perhaps you can see if you can reproduce the "cannot type" thing using my cleaned up code? I've pushed a preliminary version here: https://github.com/matthijskooijman/Arduino/tree/cleanup (branch is prone to rebasing and force pushing).

As for improving the navigation of tabs when they do not fit on the screen, #3601 is about that, so let's keep discussion about that in that ticket.

matthijskooijman, 2015-12-12:
@NicoHood , Could you see if #4279 fixes these issues for you? If you have time, a review of the code would also be welcome :-)

matthijskooijman, 2015-12-30:
@NicoHood you said that #4279 improves a bit, but doesn't fix this issue entirely. Could you see if #4363 does make it behave as expected?

NicoHood, 2016-01-02:
Same behavior as described before (after the minor improvement). I just tested to hold down the keys and it breaks again. Haven't tested anything else.

matthijskooijman, 2016-01-04:
@NicoHood thanks for checking. Interestingly enough I cannot reproduce your issue here, making this hard to fix. If you can somehow find out what happens, be sure to let us know, but for now I won't be looking into this further (it's a bit of a minor issue anyway).

NicoHood, 2016-01-04:
You might want to try elementary os in a vm real quick? It might be gnome3 related and still a problem? Even though its quite minor now, since it is fixed as long as you dont hold the buttons 2 seconds down.

matthijskooijman, 2016-01-04:
I'm already running gnome3, so that's probably not it. I found a [virtualbox image of Elementary OS](http://sourceforge.net/projects/osboxes/files/vms/vbox/elementary-OS/Freya/0.3.2/Elementary_0.3.2-64bit.7z/download), I'll try inside that. Is 1.6.7 broken for you?

NicoHood, 2016-01-06:
I am always using hourly builds, no idea if 1.6.7 has the fix applied. However the bug was always there, now its just less dramatic.

I'd just install it real quick myself in a vm. I dont think an image is required, but just do so ;)

lmihalkovic, 2016-01-07:
If you are having issues with continuously cycling through the tabs (e.g. on slower machines), a possible way to address the issue could be to only cycle through the file names until the user finally releases the ALT key. It actually works rather smoothly.

[image]

[image]

per1234, 2017-07-04:
I can reproduce the "switches 2 tabs" issue in Arduino IDE 1.6.7 but it doesn't occur in Arduino IDE 1.8.3 (Windows 7 64 bit). So I think that part of the issue is solved.

I was never able to reproduce the "cannot type anything" issue. However, I do notice that in Arduino IDE 1.6.7 when you hold ctrl+alt+arrow for a while the IDE continues switching tabs for some time after the keys are released. With Arduino IDE 1.8.3 the tab switching stops immediately after the keys are released so this indicates to me there has been some improvement of the IDE in that area since the time of this issue. 

@NicoHood, would you mind doing a quick test with the latest IDE version for the "cannot type anything" issue to see if we can close this issue now?

NicoHood, 2017-07-04:
If I open a new sketch, create 2 new tabs and press ctrl + alt + right nothing happens. Did I do it right, its been such a long time, i dont remember.

per1234, 2017-07-04:
By "nothing happens" do you mean that it doesn't switch tabs?

Since I was never able to reproduce the "cannot type anything" issue I can only guess from your description:
1. Open https://github.com/watterott/HDMI-Display/tree/master/software/HDMI-Display
2. Press and hold ctrl+alt+right for a few seconds.
3. After releasing ctrl+alt+right try to type in the Arduino IDE editor.

I wonder if the "cannot type anything" issue was caused by the continued switching of the tabs after releasing ctrl+alt+right behavior Arduino IDE ~1.6.7 had. While it's doing that the IDE is locked up but it stops for me after only a few seconds. If for some reason the duration of continued tab switching was longer on your system then that would indeed cause the "cannot type anything" issue and that should be fixed in 1.8.3 since it immediately halts tab switching.

NicoHood, 2017-07-04:
ctrl+alt+right is not functional on my 1.8.3 build on **archlinux**. It simply does nothing. I've also tested the hdmi sketch you linked. This seems to be a new issue then.

facchinm, 2017-07-04:
I'm on Arch too and the `ctrl + something` shortcuts are usually broken by some override at system level. For example, on KDE I can switch tabs with `ctrl + alt +tab` (right to left) or `ctrl + tab` (left to right) but not with right/left arrow. On Openbox, nothing happens in both the cases :smile: I'd give this issue a `wontfix` label since it only applies to a minimum user group (and doesn't really destroys the UX).

NicoHood, 2017-07-04:
I am using XFCE and i am wondering why it is overwritten. Because I've set up everything on my own and this should not happen. But this might be the reason.