arduino/Arduino #4230: Parsing Mac OS X's `system_profiler` for USB information is a blunderbuss.

mralexgray, 2015-11-30:
I constantly see `Arduino.app` spawning the process 

    -+= 00001 root /sbin/launchd
     \-+= 10895 me /.../Arduino.app/Contents/MacOS/Arduino
       \-+- 13588 me /usr/sbin/system_profiler SPUSBDataType
         \--= 13594 me /usr/sbin/system_profiler -nospawn -xml SPUSBDataType -detailLevel full

The `system_profiler` is notoriously slow, resource hungry, and certainly not the appropriate tool to query the USB subsystem's  capabilities.   It may be "easy" just to parse some XML that Apple graciously provides..  but that unfortunately does not make it prudent and/or sane to do so.

I see the `java` code calling out to this <strike>API</strike> command-line tool  in `arduino-core/src/processing/app/macosx/Platform.java` ...

```java
    try {
      CommandLine toDevicePath = CommandLine.parse("/usr/sbin/system_profiler SPUSBDataType");
      executor.execute(toDevicePath);
      return new String(baos.toByteArray());
    } catch (Throwable e) {
      return super.preListAllCandidateDevices();
    }
```
I confess to have _never even glanced_ at a piece of Java software before - nor do I hope to start now.. but I'd _imagine_ that there are plenty of (more) native bridges - between the Cocoa/ObjC subsystem and Java (or at least some way to access the hardware that isn't as brutal and archaic as parsing the output of a utility designed for only the most occasional of system inquiries).

Please let me know if this kind of integration is outside of your team's comfort zone, and I'm sure myself or some other issue onlooker can help.  I definitely think this constant polling is affecting `Arduino.app`'s performance, as well as needlessly burdening the host system as a result.  I would guess offhand that `IOKit` and/or `SystemConfiguration` frameworks would be a good place to start to find an alternative method to obtain the information you need re: the serial ports / whatnot.  If not, there are great alternatives such as [ORSerialPort](https://github.com/armadsen/ORSSerialPort) that may also be of some service in this endeavor.

ffissore, 2015-11-30:
Uhm so our current implementation is notoriously slow, resource hungry, 
brutal and archaic. Wow.

Alex Gray ha scritto il 30/11/2015 alle 03:16:
> ...but I'd /imagine/ that there are plenty of (more) native bridges...

Please let us know if you come up with an implementable solution, which 
is faster than the current one. ORSSerialPort is an ObjC library, which 
means we could use it only if you could code a JNI wrapper around it, or 
if you can make a command line version (they have a CLI demo, but it's 
interactive, so it won't work)

Looking forward for your proposals

lmihalkovic, 2015-11-30:
I guess this is the true wonders of open source, when it becomes synonymous for we-can-bestow-any-blunderous-code-on-users-and-make-it-your-responsibility-if-you-want-anything-better.

mralexgray, 2015-11-30:
@lmihalkovic, hey.. it's better than a lot of devs... who would simply close the issue!  @ffissore is a `JNI` wrapper anything like an `ffi` wrapper?  i guess I'm asking.. is a JNI blah blah blah a textual means of IPC, or some other method that hooks into standard  `C` static / shared libraries? Also,what exact information are  you getting from the `system_profiler` _now__, so I can narrow down the scope of what this would actually need to do?  PS:  Didn't mean to sound dramatic.. I just do get frustrated with this software.  Has anyone brought up the lack of a "Window" menu, lol?  I'm gonna go look now!

lmihalkovic, 2015-11-30:
@mralexgray  IMHO you speak like someone who has not read many of the closed issues yet. *better* is interesting... as a dad I try everyday to teach my 8 y/o the difference between *better-than-x* and the many shades of *good*

ffissore, 2015-12-01:
@mralexgray glad you asked.

From system_profiler we look for the device file (eg: `/dev/cu.usbmodemfa121`) and return related VID&PID. vid/pid are then used to display the name of the board (eg: Arduino/Genuino Uno).
We have a [test](https://github.com/arduino/Arduino/blob/master/app/test/processing/app/macosx/SystemProfilerParserTest.java#L52) for that parser, and some [example outputs](https://github.com/arduino/Arduino/tree/master/app/test/processing/app/macosx).

A JNI (Java Native Interface) wrapper is a way the jvm can load, introspect and use native libraries (.dll, .so). You wrap the actual native code with something the jvm can use. And while you're at it you also wrap only the functions you'll use (supposing the lib has 10 functions but you're interested in only one). Even better, you code a utility method that wraps a complex call chain, providing the jvm with a simple method like `String findVIDAndPIDOf(String serialDeviceName)`

mralexgray, 2015-12-01:
OK, [I did it](https://github.com/mralexgray/ORSSerialPort).  Hopefully someone can take it from here so I don't have to learn what a `JNI` is, lol.  So far I... 

- [x] forked `ORSerialPort`
- [x] Added a category to it, enabling USB introspection.
- [x] Made it into a dynamic library.
- [x] Created a demo app to illustrate usage.
- [ ]  Integrated into `Arduino.app`

So there is a SINGLE function to call after linking to `usbserialquery` (via `-lusbserialquery` ?)... 

    NS_INLINE char* getInfoForSerialUSB(id name);

It expects an `NSString *`, but that could be changed, or a wrapper can be created to pass it one.  It returns a `char` pointer.  There is no source file. It is implemented solely in the single header file `usbserialquery.h`.  Compiler magic, yadda yadda.  The library itself is simply a compiled `dylib` of the files included with `ORSerialPort`, so theoretically, those files could just be compiled into `Arduino.app`.. but again, don't know if that is even possible with a Java-driven app bundle.

In the demo, passing my Teensy's USB name of `/dev/tty.usbmodem935591` results in 

    000016c0_00000483

I can tweak the output if needed.  Is this OK?  How do you want it?  In a pull request?  Not sure how best to do it, you guys don't seem to use submodules.. Please advise.

ffissore, 2015-12-02:
Thank you @mralexgray ! Way to go! Best thing you can do now it make a pull request against ORSerialPort repo, so that your changes gets merged into their master, thus receiving their feedback (osx hackers), which is much more apt than ours (osx users)

I'm looking at your code to see what remains for turning it into a JNI lib

ffissore, 2015-12-02:
Two things you can improve are: don't commit binary files (they'll probably ask you to remove it anyway), don't convert VID and PID to int, use the original hex value

ffissore, 2015-12-02:
Ah, and don't append a newline, it's useless

ffissore, 2015-12-02:
One last comment: you can avoid using `break` and `continue` and write it this way
```ObjC
for (ORSSerialPort * obj in ORSSerialPortManager.sharedSerialPortManager.availablePorts) {
  if ([q isEqualToString:obj.name]) {
    info = [NSString stringWithFormat:@"%08x_%08x\n", obj.vendorID.intValue, obj.productID.intValue];
    return (char*)info.UTF8String;
  }
}
return (char*)info.UTF8String;
```
Although I don't know if ObjectiveC allows early returns

mralexgray, 2015-12-02:
@ffissore RE: merging `OR` upstream..  I don't think we need to...  the changes are all for _our_ benefit.. and my fork can be tagged and/or submodularized into `arduino-core`... or imported as a freestanding files into this repo.  I don't think my changes, _although fabulous_, necessarily are interesting to `OR`'s maintainers.

more importantly, and why I included the compiled binary,... is how can this be a drop-in replacement for you `system_profiler_shenanigans`.  are you actually going to "receive" the byte-code `char * ` return pointer, or are you going to continue to parse the textual output.  eiter way I can update the code to just return EXACTLY what you had/need so nothing else has to change.

mralexgray, 2015-12-02:
@ffissore re: "don't convert VID and PID to int, use the original hex value".. I only am providing one of many possible representations of the `VID` and `PID`.  they are provided by the category as abstract `NSNumbers`, which are a catch-all `class-cluster` that is `Objective-C`'s way of encapsulating OO "numbers".  They are whatever you want them to, whenever you want them to be that...  be it BOOL, HEX, or roman numerals, even.  In this case, I used the intValue, along with the `%8x` format string to provide a hex "representation'.

ffissore, 2015-12-02:
The goal of sending your patch upstream is to delegate to OR devs and
hackers its maintenance. We hadn't even considered using that lib because
of our lack of knowledge, do you really think we will be good enough in
maintaning it?
Plus, there's may be some dev out there that , thanks to your patch , will
male life easier to her users.
Please send your patch upstream

mralexgray, 2015-12-03:
i wrote this patch because ***THIS*** software has the problem!  `Arduino.app`'s current solution makes my (eveyones') computer slow, is the _wrong way to do it_, and can be *easily* fixed.  `ORSerialPort` is merely an implementation detail!

mralexgray, 2015-12-04:
OK, so i redid things so that NOTHING needs to change with `Arduino.app`, *except which utility to call*!
It is just a simple command line utility that gives you JUST the information previously provided by `profiler` in the SAME format.  All your previous tests should pass.

It now also ___GUESSES___(!) what your correct device is.. (if that info is not provided).

Now, running `arduino_usb_profiler` on my system (with NO arguments, even) results in...

    SLAB_USBtoUART

    Product ID: 0x10c4
    Vendor ID: 0xea60

It is a single, precompiled binary, that will work on all `10.7`+ 32/64 bit Macs, for as long as Apple continues to use Intel chipsets.  All you have to do is bundle it in, and change one line of code.  The path to `system_profiler" changes to the in-bundle path of `arduino_usb_profiler`.

Proof is in the pudding...

> `arduino_usb_profiler  0.01s user 0.01s system 26% cpu 0.047 total`

lmihalkovic, 2015-12-06:
just of thought.. but I think the forking is the least of the issues with that code... the whole polling loop structure looks [place-you-own-adjective-here]

lmihalkovic, 2015-12-06:
This afternoon I played around with usb4java (javax.usb jsr80 implementation backed by libusb). It looked promising, and would allow you to have a single codebase for all 3 platforms. The distro works fine with the model you've used for astyle (I tested with the OS X version). You might want to look in that direction.

facchinm, 2016-01-07:
Fixed by #4211

closed 2016-01-07 by facchinm