arduino/Arduino #4244: String concat reference page is incorrect

per1234, 2015-12-03:
https://www.arduino.cc/en/Reference/StringConcat
The Description, Syntax, Parameters, and Returns sections are all completely wrong. I suggest changing it to something like:

Description
Appends the parameter to the String.

Syntax
string.concat(parameter)

Parameters
parameter: Allowed types are String, string, char, byte, int, unsigned int, long, unsigned long, float, double, __FlashStringHelper(F() macro).

Returns
true: success
false: failure (in which case the string is left unchanged).

Originally reported at: http://forum.arduino.cc/index.php?topic=363080.0

agdl, 2015-12-04:
@per1234 changed as you suggested thank you!

per1234, 2015-12-04:
@agdl I think the line:
>true: success false: failure (in which case the string is left unchanged). 

should be written in two lines like:
>true: success
false: failure (in which case the string is left unchanged).

This is consistent with the way that it is done on other reference pages such as: https://www.arduino.cc/en/Reference/StringStartsWith

per1234, 2016-02-27:
@agdl the formatting error I reported in my last comment still has not been fixed.

Lauszus, 2016-02-27:
B

per1234, 2016-03-14:
@agdl https://github.com/arduino/Arduino/issues/4244#issuecomment-161943110 still not fixed,  if you don't have time to fix it now please reopen this issue so it doesn't get lost.

matthijskooijman, 2016-03-15:
@per1234, done, thanks for persisting :-)

agdl, 2016-03-16:
Sorry, actually I didn't noticed this mail :( I did it. Please take a look and close the issue

closed 2016-03-16 by per1234