arduino/Arduino #4246: How to send from slave to master more than 200 bytes?

lettierimarco, 2015-12-03:
How to send from slave to master more than 200 bytes?
I've done this but no changes:
#define TWI_BUFFER_LENGTH 300
#define RAWBUF_BYTE  TWI_BUFFER_LENGTH
#define TWI_B_OVERRIDE TWI_BUFFER_LENGTH
#define BUFFER_LENGTH TWI_BUFFER_LENGTH

Wire.begin(2);
byte toSend [300]; 
for(int i=0;i<TWI_BUFFER_LENGTH;i++);
toSend[i]=i;
Wire.write(toSend,300);

NicoHood, 2015-12-03:
Your loop does not run, you need brackets and need to remove the ;

No idea about twi itself, just looking at your code which is incorrect in general.

lettierimarco, 2015-12-03:
It's obvius that the code is for example..#define TWI_BUFFER_LENGTH 300
#define RAWBUF_BYTE TWI_BUFFER_LENGTH
#define TWI_B_OVERRIDE TWI_BUFFER_LENGTH
#define BUFFER_LENGTH TWI_BUFFER_LENGTH
Void setup(){
Wire.begin(2);
byte toSend [300]; 
for(int i=0;i<TWI_BUFFER_LENGTH;i++);
toSend[i]=i;
Wire.write(toSend,300);}
Void loop (){}

mikaelpatel, 2015-12-03:
No there is a limit of 32 bytes for the TwoWire/TWI implementation. Please see https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/libraries/Wire/Wire.h#L28.

lettierimarco, 2015-12-03:
Have you ever tried to send more bytes?
I've just tried to send 14 bytes from Arduino to raspberry. I receive only first byte and alla others like 0xFF, but if i send 9,10 bytes all works

cmaglie, 2015-12-04:
> Have you ever tried to send more bytes?

yes, a lot of users do without problems.

About your code:

1. The defines at the beginning of the sketch:
```C++
#define TWI_BUFFER_LENGTH 300
#define RAWBUF_BYTE TWI_BUFFER_LENGTH
#define TWI_B_OVERRIDE TWI_BUFFER_LENGTH
#define BUFFER_LENGTH TWI_BUFFER_LENGTH
```
seems useless, where you have find it?

2. You should put `write` between an `beginTransmission(..)` / `endTransmission(..)` block, please see the `master_writer` example of the Wire library.

3. In your for loop there is a `;` at the end of the line:
```
for(int i=0;i<TWI_BUFFER_LENGTH;i++);
```
that is equivalent to an empty loop.

4. Once you fixed the above errors, to increase the buffer size you must change the `Wire.h` file line pointed by @mikaelpatel

lettierimarco, 2015-12-04:
Begintrasmission is for master. If you look the sample code i sent i'm a slave at address 2. No begin/end trasmission are required for slave.

cmaglie, 2015-12-04:
If your arduino is acting as an I2C slave then you need to implement a `receive` event callback, please look at the `slave_receiver` / `slave_sender` examples to see how to do it (you probably need something like `slave_sender`).

@lettierimarco 
BTW this issue tracker is for keeping track of problems in the Arduino code, while this issue looks like more a request for assistance for your own code. For this kind of discussions the forum is a better place to ask -> http://forum.arduino.cc/

closed 2015-12-04 by cmaglie

lettierimarco, 2015-12-04:
I've implemented it. I wrote this code like example. I forget to put receive method. In receive method i've the wire.write but on more bytes the issue happens.

lettierimarco, 2015-12-04:
Are you sure that a slave could write more than 200 bytes in i2c? I'm correctly sending 9 bytes, trying 17 bytes I receive first byte and all oxFF.
The problem is the same doing wire.write("jsjehfishehefjfiehejrjfufehbejf"); 
I can't send big data. 
No look example code, is not a syntax error.

lettierimarco, 2016-09-15:
https://github.com/arduino/Arduino/issues/4882