arduino/Arduino #4247: SevSeg 7-segment tutorial error

gkkes, 2015-12-03:
Just a simple typo, but had me confused for a while.

http://playground.arduino.cc/Main/SevenSegmentLibrary

To install, copy the SevSeg folder into your arduino sketchbook\libraries folder. More detailed instructions are here.
-----------------------------------------------------------------------------------------------
Setting Up

 #include "SevSeg.h"
 SevSeg sevseg; //Instantiate a seven segment object
 
 void setup() {
   byte numDigits = 4;
   byte digitPins[] = {2, 3, 4, 5};
   byte segmentPins[] = {6, 7, 8, 9, 10, 11, 12, 13};
   sevseg.begin(COMMON_ANODE, numDigits, digitPins, segmentPins);
   ...

digitPins is an array that stores the arduino pin numbers that the digits are connected to. Order them from left to right.
digitPins is an array that stores the arduino pin numbers that the segments are connected to. Order them from segment a to g , then the decimal place.
---------------------------------------------------------------------------------------------------
The bottom line should read -
segmentPins is an array that stores the arduino pin numbers that the segments are connected to. Order them from segment a to g , then the decimal place".

ffissore, 2015-12-03:
Thank you for you issue. "Playground" is a community driven wiki, is not maintained by Arduino. You can edit each page on your own, including the one you found broken

closed 2015-12-03 by ffissore

per1234, 2015-12-05:
@gkkes I have made the suggested edit to the Arduino Playground page and also submitted a pull request for the updated documentation to the SevSeg repository: https://github.com/DeanIsMe/SevSeg/pull/5