arduino/Arduino #4248: IDE fails to start under Linux

RichardPillay, 2015-12-03:
This seems be the exact same problem as Issue 3365. That Issue spoke of the problem first as a MacOS issue, then as an issue of tool versions. 

I'm having the exact issue on a new installation of version 1.6.6 running under Ubuntu 15.10. Tried several ways of resolving it, then discovered issue 3365, which lead me to install multiple versions and try them. What I found was that 1.6.3 works immediately, while all later versions, including the nightly builds, exhibit the problem.

Does anyone have any ideas?

Best Regards,
Richard.

ffissore, 2015-12-03:
Can you please start the IDE from the command line and paste the output?

RichardPillay, 2015-12-04:
OK, here we go.
Arduino-1.6.3 - No Output. Graphical window starts up and all functions I've tried so far have worked normally, including compling an example sketch and uploading to a Mega.

*Output from 1.6.4, 1.6.5, 1.6.6 and Nightly build are the same. I haven't checked line-for-line, but the impression is that the output is identical.*
Arduino-1.6.6: 
richardp@Lian-Li:~/Apps/Arduino$ cd arduino-1.6.6/
richardp@Lian-Li:~/Apps/Arduino/arduino-1.6.6$ ./arduino 
Picked up JAVA_TOOL_OPTIONS: 
java.lang.VerifyError: class org.bouncycastle.asn1.ASN1Primitive overrides final method equals.(Ljava/lang/Object;)Z
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:760)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:467)
	at java.net.URLClassLoader.access$100(URLClassLoader.java:73)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:368)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:362)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:361)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:760)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:467)
	at java.net.URLClassLoader.access$100(URLClassLoader.java:73)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:368)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:362)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:361)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at org.bouncycastle.openpgp.PGPObjectFactory.nextObject(Unknown Source)
	at cc.arduino.contributions.GPGDetachedSignatureVerifier.verify(GPGDetachedSignatureVerifier.java:61)
	at cc.arduino.contributions.SignatureVerifier.isSigned(SignatureVerifier.java:46)
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:81)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:594)
	at processing.app.Base.<init>(Base.java:292)
	at processing.app.Base.guardedMain(Base.java:219)
	at processing.app.Base.main(Base.java:135)

RichardPillay, 2016-05-22:
It is now May 2016, and I've come to the conclusion today that this is not actually a bug, but an issue with something on my computer, probably my java installation.

I've upgraded to Ubuntu 16.04, and downloaded the latest 1.6.9, and had exactly the same issue. Since it is not very likely that a bug of this importance is not affecting many other Linux users, I created a virtual machine with a NEW install of 16.04 (mine is my old installation that was upgraded in place). I then installed 1.6.9 and found it works. I'm therefore closing this issue, and will do a new installation of 16.04 on my computer. I expect that will let me upgrade from 1.6.3, which I've been running since I had this issue.

My thanks to everyone who responded...

closed 2016-05-22 by RichardPillay