arduino/Arduino #4249: repeating java.lang.NullPointerException

ec3hp-de, 2015-12-03:
When I run arduino-1.6.6 this message is shown every 3 seconds in the messages panel below the sketch text:

java.lang.NullPointerException
        at javax.jmdns.impl.NetworkTopologyDiscoveryImpl.getInetAddresses(NetworkTopologyDiscoveryImpl.java:62)
        at cc.arduino.packages.discoverers.network.NetworkChecker.run(NetworkChecker.java:57)
        at java.util.TimerThread.mainLoop(Timer.java:555)
        at java.util.TimerThread.run(Timer.java:505)

Environment:
Linux 64-bit, OpenSuSE 12.3 (Virtual machine without network access)

I can upload sketches to my board via USB and they work, but it is difficult to separate any compiler error messages from the repeating java message.

ffissore, 2015-12-04:
Could it be that your computer has NO network interfaces? Not even localhost?

ffissore, 2015-12-04:
I've added a check for when there are not network interfaces. Fix will be available with next hourly build. http://www.arduino.cc/en/Main/Software#hourly Shouldn't this fix your issue, please reopen

closed 2015-12-04 by ffissore

ec3hp-de, 2015-12-04:
With the current hourly build this message is gone. Thanks.

The virtual environment was set up with no network interfaces. Typing 'ifconfig -a' shows a loopback interface, but it is not enabled.

ffissore, 2015-12-04:
Yeah, that was the cause of the error message: the running computer is expected to have localhost at least. It's actually a fault of a library we use, but I wrapped the faulty call with a check, so you can run the IDE without any active interface

commit 9a39e5e6 (2015-12-04): Some configuration may totally miss network interfaces, even localhost. Fixes #4249