arduino/Arduino #4251: Error compiling arduino code  using newewst arduino IDE

anandas, 2015-12-04:
i have issue for newest ide arduino  here is the code when trying compile an arduino code :
Arduino: 1.6.6 (Windows XP), Board: "Arduino/Genuino Uno"

C:\Program Files\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "C:\Program Files\Arduino\hardware" -tools "C:\Program Files\Arduino\tools-builder" -tools "C:\Program Files\Arduino\hardware\tools\avr" -built-in-libraries "C:\Program Files\Arduino\libraries" -libraries "C:\Documents and Settings\Owner\My Documents\Arduino\libraries" -fqbn=arduino:avr:uno -ide-version=10606 -build-path "C:\DOCUME~1\Owner\LOCALS~1\Temp\buildca8c5bbd40bceb31a8075a1e8725060f.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\DOCUME~1\Owner\LOCALS~1\Temp\untitled132479656.tmp\sketch_dec05a\sketch_dec05a.ino"
Exception 0xc000001d 0x1003f 0x0 0x503f39
PC=0x503f39

math.initÂ·1()
	c:/go/src/math/pow10.go:34 +0x19
math.init()
	c:/go/src/math/unsafe.go:21 +0x3f
strconv.init()
	c:/go/src/strconv/quote.go:455 +0x45
os/exec.init()
	c:/go/src/os/exec/lp_windows.go:123 +0x54
main.init()
	c:/jenkins/workspace/arduino-builder-windows/main.go:392 +0x49

goroutine 2 [runnable]:
runtime.forcegchelper()
	c:/go/src/runtime/proc.go:90
runtime.goexit()
	c:/go/src/runtime/asm_386.s:2287 +0x1

goroutine 3 [runnable]:
runtime.bgsweep()
	c:/go/src/runtime/mgc0.go:82
runtime.goexit()
	c:/go/src/runtime/asm_386.s:2287 +0x1
eax     0x10cfc068
ebx     0x62bd88
ecx     0x10cf2000
edx     0x62bd88
edi     0x0
esi     0x4c18
ebp     0x0
esp     0x10cfbf88
eip     0x503f39
eflags  0x10206
cs      0x1b
fs      0x3b
gs      0x0
arduino-builder returned 2

Error compiling.

ffissore, 2015-12-04:
Can you share the sketch source code?

anandas, 2015-12-04:
i used blink code from example

ffissore, 2015-12-04:
Ok, I'm looking for a windows xp box to test it. Which language is your windows?

anandas, 2015-12-04:
/*
  Blink
  Turns on an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the Uno and
  Leonardo, it is attached to digital pin 13. If you're unsure what
  pin the on-board LED is connected to on your Arduino model, check
  the documentation at http://www.arduino.cc

  This example code is in the public domain.

  modified 8 May 2014
  by Scott Fitzgerald
 */

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin 13 as an output.
  pinMode(13, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(13, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);              // wait for a second
  digitalWrite(13, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);              // wait for a second
}

anandas, 2015-12-04:
english

ffissore, 2015-12-04:
I'm sorry, I can't reproduce it. I've an almost clean windows xp and it works fine. Can you try using the latest hourly? http://www.arduino.cc/en/Main/Software#hourly Maybe some of the changes had a side effect and fixed your issue too

anandas, 2015-12-04:
still same issue..... hmm its wierd...
the old version i use ver 1.6.4 run smoothly maybe i should reversed to old version

ffissore, 2015-12-04:
What's the output when you open a command prompt, type `C:\Program Files\Arduino\arduino-builder` and press enter?

anandas, 2015-12-06:
there is no such sub folder for arduino-builder  in Arduino installer folder

koliplus, 2016-01-06:
Hi,

I have a similar problem I think.
My OS: Windows XP ; Arduino IDE: v 1.6.7 ; Board: Arduino/Genuino Uno
When I try to compile sketch I have the following error messages :

Exception 0xc000001d 0x1003f 0x0 0x508549
PC=0x508549

math.initÂ·1()
	c:/go/src/math/pow10.go:34 +0x19
math.init()
	c:/go/src/math/unsafe.go:21 +0x3f
strconv.init()
	c:/go/src/strconv/quote.go:455 +0x45
os/exec.init()
	c:/go/src/os/exec/lp_windows.go:123 +0x54
main.init()
	c:/jenkins/workspace/arduino-builder-windows/main.go:386 +0x49
eax     0x0
ebx     0x4d6de0
ecx     0x20122000
edx     0x0
edi     0x634280
esi     0x6fe30
ebp     0x6344a0
esp     0x2012bf88
eip     0x508549
eflags  0x10206
cs      0x1b
fs      0x3b
gs      0x0
arduino-builder returned 2

Error compiling.

I try to compile a "blink" example. When I change a board type I have similar error.
When I try to execute arduino-builder from Windows command line I have the same effect.
C:\>m:\arduino-1.6.7\arduino-builder.exe
Exception 0xc000001d 0x1003f 0x0 0x508549
PC=0x508549

math.init┬Ě1()
        c:/go/src/math/pow10.go:34 +0x19
math.init()
        c:/go/src/math/unsafe.go:21 +0x3f
strconv.init()
        c:/go/src/strconv/quote.go:455 +0x45
os/exec.init()
        c:/go/src/os/exec/lp_windows.go:123 +0x54
main.init()
        c:/jenkins/workspace/arduino-builder-windows/main.go:386 +0x49

goroutine 2 [runnable]:
runtime.forcegchelper()
        c:/go/src/runtime/proc.go:90
runtime.goexit()
        c:/go/src/runtime/asm_386.s:2287 +0x1

goroutine 3 [runnable]:
runtime.bgsweep()
        c:/go/src/runtime/mgc0.go:82
runtime.goexit()
        c:/go/src/runtime/asm_386.s:2287 +0x1
eax     0x2012c068
ebx     0x633d88
ecx     0x20122000
edx     0x633d88
edi     0x0
esi     0x4e48
ebp     0x0
esp     0x2012bf88
eip     0x508549
eflags  0x10206
cs      0x1b
fs      0x3b
gs      0x0

This error appears on Arduino IDE v 1.6.6 too, I think this error will appear on every IDE versions with arduino-build.exe (on IDE v 1.6.5 compilation is successful).

I wonder what kind of strange characters are close to the code:
math.init**Â·1**()
	c:/go/src/math/pow10.go:34 +0x19
Strange. It isn't ?

lulol, 2016-02-04:
Also found this issue with v1.6.6 and v1.6.7, but v1.6.4 and v1.6.5-r5 both work fine in Windows XP SP3.

I'm testing them with an old AMD Athlon XP+ CPU. May be is the lack of SSE2 support in these CPUs, which is enabled by default in recent compilers that must be explicitely configured to support old CPUs?

Running arduino-builder.exe in these two newer versions fails similarly to koliplus' log but the CPU register dump is different.

anandas, 2016-02-05:
Yes i think that issue is SSE2 unsupported Processor any idea to fix it ?

cmaglie, 2016-02-05:
Here some other threads that may be related:

https://github.com/golang/go/issues/11631
https://github.com/syncthing/syncthing-inotify/issues/58

It seems that we should compile the arduino-builder with `GO386=387` option enabled to disallow the use of SSE2 instructions, I will have a look at this one.

cmaglie, 2016-02-11:
@anandas @koliplus @lulol 
can you try the latest hourly build?
We have applied the `GO386=387` flag and it seems that no SSE2 instructions are used in the builder now, but we have no way to test it.

lulol, 2016-02-11:
I can confirm that the latest arduino-builder.exe v1.3.11 (md5:0d7b426e24c92c24dbf01335451b1650) in the nightly build (md5:e50c906949d5446941efc3a0199127df) runs without crashing in an Athlon XP which doesn't suport SSE2.

koliplus, 2016-02-14:
I confirm, the latest arduino-builder.exe v1.3.12 works without errors on Athlon XP 2600+ on Windows XP and on Intel Pentium M 1.7 on Ubuntu 14.04
- replaced in place of the original in Arduino IDE v.1.6.7

cmaglie, 2016-02-17:
Thanks! The patched builder will be available from the release 1.6.8.

closed 2016-02-17 by cmaglie

anandas, 2016-02-20:
thanks its working

mrchadwickusa, 2016-03-22:
I had the same problem with the Linux release on Puppy Linux slacko 5.7 on an Athlon XP 2800. Doesn't seem to be fixed in the latest Linux release version 1.6.8.

facchinm, 2016-03-22:
Uh, we only applied the fix to Windows build... 
You are the first reporting that Linux 386 version suffers it too... 
This will surely be solved in the next builder release

mrchadwickusa, 2016-03-22:
Thank you for the very fast reply!
I want to use this older machine with linux to do development on the bench, and Arduino is of course part of that.  I will also try the windows release of 1.6.8 on Win XP with this machine.
Thanks again!

AjuAJ, 2017-02-06:
SOLVED!!!!!   Just run the arduino ide as administrator.

buenavist, 2017-08-20:
Hi readers,

I am having troubles using an old computer with some "new" version of arduino (1.6.12 but I am open for other version).
The troubles looks very similar to above as I am on Linux.
I met similar issues with Skype that do not support SS3 anymore (a big fail for this software that I do not use anymore). My CPU is a transmetta crusoe (dated 2002 from the top of my head, so quite old).

If I can help fighting unecessary throwing old computer in the bin without having a descent reason.

Best regards,