arduino/Arduino #4253: Upgrading to Arduino 1.6.6 Breaks YunClient

azdle, 2015-12-04:
My library that uses YunClient in some of the examples no longer compiles under 1.6.6.

Here is the error:

    ExoYunGettingStarted:63: error: using typedef-name 'YunClient' after 'class'
     class YunClient client;
           ^
    In file included from /Users/patrick/Documents/Arduino/libraries/Exosite/examples/ExoYunGettingStarted/ExoYunGettingStarted.ino:44:0:
    /Users/patrick/Downloads/Arduino.app/Contents/Java/libraries/Bridge/src/YunClient.h:25:22: note: 'YunClient' has a previous declaration here
     typedef BridgeClient YunClient;
                          ^

Here is the code that is getting the error: 
https://github.com/exosite-garage/arduino_exosite_library/blob/bf6fe8c0b14b0dd4a05cba6a9cbae679f2a30658/examples/ExoYunGettingStarted/ExoYunGettingStarted.ino#L63-L64

It seems that #2490 introduced a non-backwards compatible change.

cmaglie, 2015-12-04:
To create a `YunClient` you must write:

```C++
YunClient client;
```
and not
```C++
class YunClient client;
```

The `class` keyword is used to define a new user defined class.

> It seems that #2490 introduced a non-backwards compatible change.

The change should be 100% backward compatible (BTW I cannot exclude that some very esoteric cases may create problems).

azdle, 2015-12-04:
Heh, I guess I need to stop pretending that knowing C means that I understand C++ too.

Thanks for that.

closed 2015-12-04 by azdle