arduino/Arduino #4258: Serial Monitor window not appearing

erdemoncel, 2015-12-05:
I have Arduino version 1.6.6 and when I try to open serial monitor nothings happen. 

I think it works but no visible window. I use Ubuntu 12.04.

ffissore, 2015-12-07:
Are you using a multimonitor setup? Sometimes windows get lost in the workspace and the only way (I found) to get them back is expanding the workspaces view and drag them to their correct location

erdemoncel, 2015-12-08:
No. I use only one monitor.

After removing and installing display driver, I'd be able to open serial monitor.

Also when starting Arduino software splash screen appears from right. But I suspect this may be an OS related bug. xdpyinfo  | grep dimensions reports screen dimension as 2304x1024 pixels but in fact it is 1280x1024 pixels.

Thanks..

closed 2015-12-08 by erdemoncel

robo-shark, 2016-10-31:
lowering screen resolution from 1920x1080 to 1366x768 then running serial monitor made it appear, then I just changed back the resolution to the way it was and everything was OK