arduino/Arduino #4260: Incorrect Diagram https://www.arduino.cc/en/Tutorial/ReadASCIIString

vgatesr, 2015-12-05:
Your instruction show connecting the anode to +5 v and it does not work. Move the connection to GND and it does.
This one works (https://www.bing.com/images/search?q=rgb+led+arduino&view=detailv2&qpvt=rgb+led+arduino&id=72BAA7D024767F034FD516F082019FAC21C64EA6&selectedIndex=0&ccid=LN28wCQI&simid=608002808251027198&thid=OIP.M2cddbcc024082add7d2374cdadb8aeb6H0&ajaxhist=0)
Yours does not.

matthijskooijman, 2015-12-05:
Uh, but the anode is supposed to get the positive voltage, and the cathode the negative one. From what you write, I suspect that you have a common cathode RGB led (one cathode, three anodes), which indeed needs to be connected differently. Since that's a different part, I think the instructions are not incorrect, but perhaps mentioning the wiring needed for a common cathode led would be a nice addition.

vgatesr, 2015-12-05:
You must be correct. I just used the RGB that was in the Stater Kit
assuming it would be consistent with the Arduino projects. Assumptions.
The starter book does not list the 4 pin RGB or any RGB for that matter. I
think I have noticed that the Ardunio community doesn't do a very good job
of listing component electrical characteristics ie data sheets.
Thanks for the response.

per1234, 2015-12-06:
>the Ardunio community doesn't do a very good job of listing component electrical characteristics ie data sheets.

The datasheet for the RGB LED is provided on the [Arduino Starter Kit page](https://www.arduino.cc/en/Main/ArduinoStarterKit):
https://www.arduino.cc/documents/datasheets/LED(RGB).pdf
As you can see it states that it's a common cathode.

vgatesr, 2015-12-06:
Right again although it does not appear in the book that came with the kit. Thanks, I am learning a lot from this.

 

Thanks,

 

Vic

agdl, 2015-12-14:
Image fixed. Thank you!

closed 2015-12-14 by agdl

leonardoemarq, 2016-03-13:
Hello, as Arturo (agdl) mentioned, the image was fixed to make it correspondent with a Common Cathode circuit (thank you Arturo). But the text stills indicating the connections as if you are using a Common Anode RGB LED. So, now the text and the image are inconsistent!

agdl, 2016-03-14:
@leonardoemarq done thank you!