arduino/Arduino #4262: Install errors from ide 1.6.6

w8one, 2015-12-06:
I got these errors installing 1.6.6 in windows 10. In all cases i chose Ignore. After install i checked and all the files were there.

c:\Program Files (x86)\Arduino\java\bin\awt.dll ---error opening file for writing  ---ignore
c:\Program Files (x86)\Arduino\java\bin\dcpr.dll ---error opening file for writing  ---ignore
c:\Program Files (x86)\Arduino\java\bin\fontmanager.dll ---error opening file for writing  ---ignore
c:\Program Files (x86)\Arduino\java\bin\java.dll ---error opening file for writing  ---ignore
c:\Program Files (x86)\Arduino\java\bin\javaw.exe ---error opening file for writing  ---ignore
c:\Program Files (x86)\Arduino\java\bin\jpeg.dll ---error opening file for writing  ---ignore
c:\Program Files (x86)\Arduino\java\bin\msvcr100.dll ---error opening file for writing  ---ignore
c:\Program Files (x86)\Arduino\java\bin\net.dll ---error opening file for writing  ---ignore
c:\Program Files (x86)\Arduino\java\bin\nio.dll ---error opening file for writing  ---ignore
c:\Program Files (x86)\Arduino\java\bin\splashscreen.dll ---error opening file for writing  ---ignore
c:\Program Files (x86)\Arduino\java\bin\t2k.dll ---error opening file for writing  ---ignore
c:\Program Files (x86)\Arduino\java\bin\verify.dll ---error opening file for writing  ---ignore
c:\Program Files (x86)\Arduino\java\bin\zip.dll ---error opening file for writing  ---ignore

c:\Program Files (x86)\Arduino\java\bin\client\jvm.dll ---error opening file for writing  ---ignore

c:\Program Files (x86)\Arduino\lib\AStylej.dll ---error opening file for writing  ---ignore

If you need more info just ask.

willy - w8one

w8one, 2015-12-06:
Apparently the uninstaller was not able to delete the files above and did not report that. I manually deleted them and reinstalled 1.6.6 and its all fine.

closed 2015-12-06 by w8one

cmaglie, 2015-12-07:
This happens when you try to uninstall with the Arduino IDE opened (that keeps these files locked). The un-installer for the next version 1.6.7 will have a check for this condition.