arduino/Arduino #4264: Modulo Library in Library Manager

pixelpixi, 2015-12-06:
Hello. You previously added my arduino library for modulo to the library manager ( https://github.com/arduino/Arduino/issues/4059 ), but it doesn't seem to be there anymore. Did it disappear? Or maybe there's something wrong with my github repo?

Thank you,
Erin

ffissore, 2015-12-07:
It seems like there's a bug in our publishing job. I'm working on it

ffissore, 2015-12-07:
Fixed. Your modulo lib is now listed again. However, it won't work. Yo have a symlink from src to firmware, but zip doesn't support symlinks and even it did, windows doesn't support symlinks. Please rename your firmware folder to src and make a new release

closed 2015-12-07 by ffissore

pixelpixi, 2015-12-08:
@ffissore: Thanks for listing it again. I've removed the symlink and it's working now. (It was an attempt to make the library work both for Arduino and Particle, but I guess I have to use separate repos).

Do you know why the library was removed? Was it because of the symlink? Or perhaps due to lack of use? We're shipping our product this week (yay!) so I expect people to start using it soon.

Thanks,
Erin

ffissore, 2015-12-08:
That symlink screwed our publishing job, and bad things happened. It was a bug, and it's now fixed